realsenseai/librealsense #862: Saving the point cloud to a .ply file using the Python wrapper

felix-tracxpoint, 2017-11-29:
| Required Info  |   |
|---|---|
| Camera Model | D400 | 
| Firmware Version | SDK 2.0  | 
| Operating System & Version | Windows 10  |
| Kernel Version (Linux Only)  |   |

I would like to save the point clouds to .ply files using the Python wrapper of the SDK, in the same way that I can save .ply files from the SDK's GUI. But I am unable to find documentation explaining how to achieve this. All I found is the following line:

`        points = pc.calculate(depth_frame)
`

But what sort of object is 'points'? What methods can one call for it? Where can I read about it?

Thanks in advance!

dorodnic, 2017-11-29:
Hi @felix-tracxpoint 
At the moment PLY conversion is not part of the core SDK but only the UI. 
We will consider adding it to the API (making it available to all the wrappers) in one of the future releases.
For now, the PLY format is rather simple, and you can use the following code as reference:
[model-views.cpp:199-289](https://github.com/IntelRealSense/librealsense/blob/master/common/model-views.cpp#L199-L289)

felix-tracxpoint, 2017-11-30:
Is there some other format I can export to, without implementing the whole thing by myself?

Thanks!

dorodnic, 2017-11-30:
There is no export functionality at the moment. If all you need is to serialize the data to save it for later use, you might want to check [SDK recording capabilities](https://github.com/IntelRealSense/librealsense/blob/master/src/media/readme.md). However, `ros-bag` format is fairly proprietary and will not help you export the 3D model into 3rd party software.

dorodnic, 2017-12-24:
You can now save MeshLab-compatible PLY files directly from pyrealsense

closed 2017-12-24 by dorodnic

felix-tracxpoint, 2017-12-25:
This is great! But how...?

dorodnic, 2017-12-25:
We added an example - [export_ply_example.py](https://github.com/IntelRealSense/librealsense/blob/development/wrappers/python/examples/export_ply_example.py). 
At the moment, it is still only on the development branch, but it will be merged to master with next release (and also added to next binary installer)

nathanlem1, 2018-02-26:
Is there any way to render (visualize) the pointcloud obtained from Intel RealSense API in Python? There are examples for rendering in C++ but not in Python. I hope you may give me a hint here. Thanks