realsenseai/librealsense #866: Read device temperature

Panepo, 2017-11-30:
| Required Info  |   |
|---|---|
| Camera Model | D410 | 
| Firmware Version | 05.06.09.00  | 
| Operating System & Version | Win10 (Build 16299)  |
| Kernel Version (Linux Only)  |   |

Hi there,
I am trying to read both ASIC and projector temperature by following:
```
import pyrealsense2 as rs
import numpy as np
import cv2

pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

cfg = pipeline.start(config)
dev = cfg.get_device()
depth_sensor = dev.first_depth_sensor()

try:
    while True:

        # Wait for a coherent pair of frames: depth and color
        frames = pipeline.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()
        
        # Convert images to numpy arrays
        depth_image = np.asanyarray(depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())

        # Apply colormap on depth image (image must be converted to 8-bit per pixel first)
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, None, 0.5, 0), cv2.COLORMAP_JET)

        # Stack both images horizontally
        images = np.hstack((color_image, depth_colormap))
        
        # Get device temperature
        tempProj = depth_sensor.get_option(rs.option.projector_temperature)
        print("tempProj: " + str(tempProj))
        tempAsic = depth_sensor.get_option(rs.option.asic_temperature)
        print("tempAsic: " + str(tempAsic ))

        # Show images
        cv2.namedWindow('RealSense', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('RealSense', images)
        cv2.waitKey(1)

finally:

    # Stop streaming
    pipeline.stop()
```
But the temperature outputs are the same while I streaming, the outputs only change while I restart the program. How to read the device temperature dynamically?
Thanks

dorodnic, 2017-11-30:
Hi @Panepo 
You are doing everything correctly. We are able to reproduce the problem but can't quite localize what is causing it. We see that when running in a simple loop (either using pipeline or using context/sensor API) often we get temperature data that is marked invalid - [ds5-options.cpp:82](https://github.com/IntelRealSense/librealsense/blob/master/src/ds5/ds5-options.cpp#L82):
```cpp
if (0 == temperature_data.*is_valid_field)
            LOG_ERROR(rs2_option_to_string(_option) << " value is not valid!");
```
This doesn't seem to happen in the Viewer, but we couldn't quite put the finger on what in the Viewer is making the firmware send valid temperature. 
As an additional data-point, the issue doesn't seem to reproduce on Linux.
We will keep looking into it and update if we find anything.

xtrawurst, 2018-01-05:
Is this issue still present in the latest 2.9.0 release of librealsense?
We also need access to reliable temperature values for our use-case.

dorodnic, 2018-01-13:
Hi @xtrawurst 
Yes, the bug is still present unfortunately. After spending quite a long time on this issue, I can only infer that it is some kind of race-condition at the firmware level. The reproduces most of the time in all applications but the viewer, but it sometimes also reproduces in the viewer.
I think I can safely say it has nothing to do with intrinsics / extrinsics calls, pipeline vs sensor API, other options read, or anything else I could think of. 
We'll try to find a work-around.

dorodnic, 2018-01-13:
Looking at some internal documentation, it seems you can read projector temperature using following commands:
```cpp
rs2::pipeline p;
auto profile = p.start();
auto dev = profile.get_device();

auto dbg = dev.as<rs2::debug_protocol>();
std::vector<uint8_t> cmd = { 0x14, 0, 0xab, 0xcd, 0x2a, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0 };

auto res = dbg.send_and_receive_raw_data(cmd);
int temp = res[4];
std::cout << temp << std::endl;
```
This method is unaffected by the mysterious `is_valid` bit on the temperature XU (the option bug).
There are two problems with this work-around:
1. I couldn't find similar command to read the ASIC temperature
2. This is not official API and therefore can break in the future

RealSense-Customer-Engineering, 2018-02-27:
[Realsense Customer Engineering Team Comment] 
Hi @xtrawurst,

would like to know if you still need help with this issue?

 Regards

xtrawurst, 2018-02-27:
@RealSense-Customer-Engineering 
I'd like to know whether the issue persists. The workaround outlined by @dorodnic has got some problems too (as described).

RealSense-Customer-Engineering, 2018-05-09:
[Realsense Customer Engineering Team Comment]
hi @xtrawurst,

we are still fixing this, will update you once we finished.

marinerain, 2018-06-27:
@RealSense-Customer-Engineering we are running into this problem too and would like to know if you found a better fix than the workaround specified above. We need both Asic and Projector temperature logs and the current fix is introducing significant conflicts.

Required Info |  
-- | --
Camera Model | D435
Firmware Version | 05.08.15.00
Operating System & Version | Ubuntu 16.04
Kernel Version (Linux Only) |  4.4.38

dorodnic, 2018-07-02:
This seems to be a somewhat tricky bug. We revisited it several times since it was originally reported, but with no luck so far, unfortunately. We were planning to revisit it **again** this sprint and update on any findings.

jeremySch, 2018-09-19:
I've been having a similar issue reading out the temperatures every frame.  As a workaround I've switched to reading out the temperature every 10 seconds and it appears to be working now.

rupak-d, 2018-10-04:
Hi @dorodnic, @RealSense-Customer-Engineering ,
Any updates on this issue?

Thank you.

RealSense-Customer-Engineering, 2018-10-05:
[Realsense Customer Engineering Team Comment]
Hi @rupak-d,

You can use the workaround code from @dorodnic first to read the projector temperature.
@dordnic, do you have more info about the fix?

dorodnic, 2018-10-07:
Still under investigation at firmware level

marinerain, 2018-12-18:
> [Realsense Customer Engineering Team Comment]
> Please try the firmware 5.10.13.0 to see if the error still can be
> reproduced.

Hi Sergey,

Thank you for the update. I transmitted that to our team and we will test
it after Christmas break.

Regards

RealSense-Customer-Engineering, 2019-01-16:
[Realsense Customer Engineering Team Comment]
Hello @marinerain,

any test update?

dorodnic, 2019-01-22:
Hopefully resolved now, @marinerain update this issue or feel free to open a new one, if the problem reproduces

closed 2019-01-22 by dorodnic

Kaju-Bubanja, 2019-08-30:
This issue still persists in the newest firmware 05.11.06.250

andre-ryll, 2019-09-03:
I can second this. Running librealsense 2.26.0 and FW 05.11.06.250 on a D435.

Interestingly, the ASIC and projector temperature options appear in the list of supported options (sensor.get_supported_options()), but if I query them individually (sensor.supports(..)) it returns false (not supported). Actually trying to retrieve the value fails with an exception (not supported option).

We need this feature as well to monitor temperatures during operation. I will try the debug protocol in the meantime but this should really get fixed.

andre-ryll, 2019-09-18:
I am just using the debug protocol snippet @dorodnic posted above to read the projector temperature now. I am still hoping for librealsense team to fix the issue with reading the option values.

nawara72, 2020-01-30:
Hi  Any updates on the temperature fix

diplodocuslongus, 2023-06-27:
@jrecasens are you streaming frames when you read the temperature? If not you will not get anything. Try the code from @Panepo at the very top.
I can read both projector and asic temperature with his code on a D455 (which, if I'm correct, has the same internal HW as the D457).