realsenseai/librealsense #867: EnhancedPhoto model

Panepo, 2017-11-30:
| Required Info  |   |
|---|---|
| Camera Model | D400 | 
| Firmware Version | 05.06.09.00  | 
| Operating System & Version | Win10 (Build 16299)  |
| Kernel Version (Linux Only)  |   |

Hi there,
There was a feature MeasureDistance in model EnhancedPhoto in SDK 1.0. 
Is there a similar feature in SDK 2.0?
Thanks

dorodnic, 2017-11-30:
At the moment there is no such functionality in the SDK. 
Unfortunately, all existing computer-vision middlewares that used to ship with RealSense SDK has been deprecated. librealsense teams goal is to offer decent **open-source** and **cross-platform** alternatives over time, using existing 3rd party libraries.

Panepo, 2017-12-04:
Hi @dorodnic 
Is there any information that I can build this function **MeasureDistance** by myself?
I think I will need an equation of the actual distance **X** of two adjacent pixels of the same depth value **Y** in depth image.

dorodnic, 2017-12-15:
Hi @Panepo 
You can always de-project depth pixels into (x,y,z) form in meters (using camera extrinsics and utility functions from rs2_utils.h) and then use euclidian distance. 
We were considering adding such example, but there are couple of complications to consider, like pixels with missing (zero) depth.

Panepo, 2017-12-18:
Hi @dorodnic 
I think I should use this function: rs2_deproject_pixel_to_point, but I don't know how to get intrinsics.
Is there any example how to get intrinsics in python?
Thanks

zivsha, 2017-12-19:
Hi @Panepo ,

In order to get the intrinsics you need to first get the profile for which you want the intrinsics.
The profile (or "stream profile") can be retrieved from the sensor, from a frame, or from the a pipeline profile.

Here's a small snippet in python that fetches the intrinsics of an infrared stream:
```py
import pyrealsense2 as rs
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.infrared)
profile = pipeline.start(config)
ir_stream = profile.get_stream(rs.stream.infrared)
ir_video_stream = ir_stream.as_video_stream_profile()
intrin = ir_video_stream.get_intrinsics()
print "Intrinsics:\n  width:" , intrin.width, "\n  height:" , intrin.height, "\n  ppx:" , intrin.ppx, "\n  ppy:" , intrin.ppy, "\n  fx:" , intrin.fx, "\n  fy:" , intrin.fy, "\n  model:" , intrin.model
```

dorodnic, 2018-01-18:
In this release we have added [distance measure example](https://github.com/IntelRealSense/librealsense/tree/development/examples/measure)
Hopefully it will help provide the code components you need.

closed 2018-01-23 by dorodnic