realsenseai/librealsense #869: How to get the intrinsic parameters with the Python wrapper?

felix-tracxpoint, 2017-11-30:
| Required Info  |   |
|---|---|
| Camera Model | D400 | 
| Firmware Version | SDK 2.0   | 
| Operating System & Version | Windows 10  |
| Kernel Version (Linux Only)  |   |

I presume it's something obvious I had overlooked.

dorodnic, 2017-11-30:
Hi @felix-tracxpoint 
Unfortunately, it is not as obvious as we would like:
```python
cfg = pipeline.start() # Start pipeline and get the configuration it found
profile = cfg.get_stream(rs.stream.depth) # Fetch stream profile for depth stream
intr = profile.as_video_stream_profile().get_intrinsics() # Downcast to video_stream_profile and fetch intrinsics
print intr.ppx
```
We are trying to keep the SDK open to non-video extensions in the future, that's why a downcast (to `video_stream_profile`) is required before you can fetch the intrinsics.

felix-tracxpoint, 2017-11-30:
Thanks, this is very helpful!

closed 2017-11-30 by felix-tracxpoint

erikcummins, 2019-08-08:
Has this been improved since this issue was created? Or do we still need to state a pipeline to retrieve the intrinsics from the camera?

lramati, 2019-08-20:
you no longer need to create a pipeline. You can query devices from a context object and then query it's sensors, or just query all available sensors directly from the context object. From here you can grab a list of all supported stream profiles, and use the same logic as above to get the intrinsics for each