realsenseai/librealsense #870: Make pyhton wrapper working

Roios, 2017-11-30:
Hello,

It can be a noob question but how can I install the python wrapper?
I other words... what do I need to do to be able to import it to my program?
Yes, basic noob here.

Thank you for your help

| Required Info  |   |
|---|---|
| Camera Model |  SR300  | 
| Firmware Version |   | 
| Operating System & Version |  Windows 10  |

dorodnic, 2017-11-30:
Hi @Roios 
If you build the SDK from source, enable `BUILD_PYTHON_BINDINGS` flag in `cmake-gui` and build `pyrealsense2` project (under `Wrappers` > `python`). The resulting binary will be under `Debug` / `Release` folder, named `pyrealsense2.pyd`. Copy it (and `realsense2.so` unless you disable `BUILD_SHARED_LIBS`) near your python script.
If you are using the [installer](https://github.com/IntelRealSense/librealsense/releases/download/v2.8.2/Intel.RealSense.SDK.exe), copy `pyrealsense2.pyd` and `realsense2.dll` from `C:\Program Files (x86)\Intel RealSense SDK 2.0\bin\x86` to your script folder. 
Then proceed to `import pyrealsense2 as rs` in your python script. 
Feel free to ask questions if you encounter any further problems :)

Roios, 2017-11-30:
Hi @dorodnic 

I installed using the installer and copied the files as you mentioned.
I have this basic script:

`try:
    import pyrealsense2 as rs
    print('ok')
except:
    print('not ok')`

The result is always 'not ok'. It can't import.
Any idea?

dorodnic, 2017-11-30:
Perhaps it is platform mismatch? With the installer we provide both x86 and x64 binaries. What error do you get if you don't wrap the thing in try?

Roios, 2017-11-30:
I tried with bot x86 and x64
I get this error: 

>  File "hello.py", line 1, in <module>
>     import pyrealsense2 as rs
> ImportError: DLL load failed: %1 is not a valid Win32 application.

As a noob, this is a mistery for me

dorodnic, 2017-11-30:
Are both `pyrealsense2.pyd` and `realsense2.dll` being copied next to `hello.py`? How do you run the script? Can you try `File` > `Open Powershell` / `Open Command-Line` and type `python hello.py`?
I'm also noticing now that `pyrealsense2.pyd` is lacking digital signature that exists on all other DLLs. Please check if there is an `Unblock` checkbox under PYD file properties. 
Other steps to troubleshoot the issue might include inspecting the PYD & DLL using the [dependency walker](http://www.dependencywalker.com/) to understand what dependencies might be missing.

Roios, 2017-11-30:
> Are both pyrealsense2.pyd and realsense2.dll being copied next to hello.py?
Yes they are, the 3 files in the same folder.

> Can you try File > Open Powershell / Open Command-Line and type python hello.py?
That is exactly how I did it.

> Please check if there is an Unblock checkbox under PYD file properties.
No, nothing like that.

> Other steps to troubleshoot the issue might include inspecting the PYD & DLL using the dependency walker to understand what dependencies might be missing.
OK, I will see what I can do from there...

Thank you very much @dorodnic

Roios, 2017-11-30:
@dorodnic I solved it.
After many try and error I installed python2.7 32 - bit and it worked.
Now I wonder, maybe this doesn't work for python 3.*!
Am I right?

dorodnic, 2017-11-30:
I was considering to bring it up as an option earlier today, but it worked for me even after uninstalling 2.7 and installing 3.6. I think it's because I was building the PYD from source with 3.6 installed. 

The wrapper is not limited to a single python version, but it seems that the precompiled binaries are built with 2.7. We will check if it is possible to include both python2 and python3 versions in the installer.

Roios, 2017-12-01:
But if someone could explain me how to build the PYD by myself using cmake....I would be super grateful!
I tried to follow the instructions and using cmake gui but in the end I never can find the PYD file. Just cpp stuff.
If someone needs to do good actions for christmas....I'm here.

Thank you all.

dorodnic, 2017-12-01:
Once you generate the project using `cmake`, you need to build everything using Visual Studio with C++ support. It is a bit cumbersome if all you need is the python module, that's why we publish precompiled binaries, but you can give it a try with [Visual Studio Community edition](https://www.visualstudio.com/downloads/)

closed 2017-12-01 by dorodnic