realsenseai/librealsense #872: Color stream crashes C++ Tutorials 'streams' app

c4stan, 2017-12-01:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version | Not sure how to check that. | 
| Operating System & Version | Win10 Version 1703 Build 15063.726 |

If I leave the color stream enabled I get the following error.
> rs::error was thrown when calling rs_start_source(device:000001B0CC51DFE0, source:VIDEO):
>     no matching media type for  pixel format 59555932
The depth stream works fine.

dorodnic, 2017-12-03:
Hi @c4stan 
I was able to reproduce this problem with the R200, and it disappeared after uninstalling and re-installing the driver for `Intel(R) RealSense(TM) 3D Camera (R200) RGB` under Imaging Devices. 
Rest of devices worked properly, including on Win 10 1703. 
Please try re-installing the driver.

dorodnic, 2017-12-15:
Hi @c4stan, could you please update if re-installing the driver helped?

c4stan, 2017-12-15:
Thank you @dorodnic for the help. 
Re-installing the driver did indeed solve the issue.
Closing.

closed 2017-12-15 by c4stan