realsenseai/librealsense #874: Error-realsense-GLFW_INCLUDE_DIR

ljc19800331, 2017-12-01:
I am using clion and try to build the librealsense code and got the following error. I am using windows 10. How can I install the glfw package in windows 10? Than you.

C:\Users\gm143\.CLion2017.3\system\cygwin_cmake\bin\cmake.exe -DCMAKE_BUILD_TYPE=Debug -G "CodeBlocks - Unix Makefiles" /cygdrive/c/Users/gm143/CLionProjects/librealsense
-- Info: REALSENSE_VERSION_STRING=2.8.2
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
GLFW_INCLUDE_DIR
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/capture
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/capture
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/capture
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/capture
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/capture
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/capture
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/capture
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/save-to-disk
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/save-to-disk
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/save-to-disk
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/save-to-disk
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/save-to-disk
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/save-to-disk
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/save-to-disk
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/multicam
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/multicam
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/multicam
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/multicam
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/multicam
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/multicam
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/multicam
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/pointcloud
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/pointcloud
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/pointcloud
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/pointcloud
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/pointcloud
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/pointcloud
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/pointcloud
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/align
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/align
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/align
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/align
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/align
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/align
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/align
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/sensor-control
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/sensor-control
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/sensor-control
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/sensor-control
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/sensor-control
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/sensor-control
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/sensor-control
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/C/depth
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/C/depth
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/C/depth
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/C/depth
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/C/depth
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/C/depth
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/C/depth
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/C/color
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/C/color
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/C/color
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/C/color
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/C/color
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/C/color
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/C/color
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/C/distance
[... 50 lines cut ...]
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/tools/depth-quality
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/tools/depth-quality
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/tools/depth-quality
   used as include directory in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/tools/depth-quality
GLFW_LIBRARIES
    linked by target "rs-capture" in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/capture
    linked by target "rs-save-to-disk" in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/save-to-disk
    linked by target "rs-multicam" in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/examples/multicam
[... cut ...]
    linked by target "realsense-viewer" in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/tools/realsense-viewer
    linked by target "rs-data-collect" in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/tools/data-collect
    linked by target "rs-depth-quality" in directory /cygdrive/c/Users/gm143/CLionProjects/librealsense/tools/depth-quality

-- Configuring incomplete, errors occurred!
See also "/cygdrive/c/Users/gm143/CLionProjects/librealsense/cmake-build-debug/CMakeFiles/CMakeOutput.log".

[Failed to reload]

zivsha, 2017-12-03:
Hi @ljc19800331 ,

Currently we only test our build against MSVC (Visual Studio) but not with `mingw` nor `cygwin` so I can't say this is 100% not an issue with our CMake but it looks to me like something is wrong with your build configuration.
If you build under windows, the `glfw` library is not searched for by CMake (`find_library`), so I fail to see how you would get the `NOTFOUND` error for this library. 
Can you make sure that your CMake enters one of the `if(WIN32)` conditions?

ljc19800331, 2017-12-03:
Great thanks. I know mingw or cygwin is not a standard installation method and I will try Visual Studio today and try to reinstall all the things again.

zivsha, 2017-12-06:
Any updates @ljc19800331 ?

alexge233, 2017-12-08:
You need [GLFW3](https://github.com/glfw/glfw) built and installed, the README or instructions do not specify it, and interestingly enough neither does the `CMakeLists.txt`. If you disable `-DBUILD_GRAPHICAL_EXAMPLES=Off` then the issue goes away.

ljc19800331, 2017-12-08:
Thank you for you guys' answers.

zivsha, 2017-12-09:
> CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
GLFW_INCLUDE_DIR

This issue simply states that CMake cannot find GLFW. If you only built the source of GLFW it is not enough, you need to install it - either with `make install` or using `sudo apt-get install libglfw3-dev` (as stated in the [installation manual](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md).

If you set the CMake flags to not build examples you shouldn't have gotten this message though. We will look into that.

zivsha, 2018-02-03:
@ljc19800331 are you still seeing this issue?

ljc19800331, 2018-02-03:
@zivsha  I have successfully installed in linux. Sorry for the late update. Thanks for the help.

For the windows 10, I will eventually find a solution to install in that OS. I will update the error if needed.

Thank you very much.

zivsha, 2018-02-04:
What is the issue with Win10?

closed 2018-02-05 by ljc19800331