realsenseai/librealsense #875: Will Raspberry pi 3 support sr300 camera

ankit-shinde, 2017-12-04:
| Camera Model | SR300 | 
| Firmware Version | Raspbian os jessie  | 

Hi,

I just want to know whether the current version will support for Raspberry pi 3 model b for SR300 camera.

Thanks :)

dorodnic, 2017-12-04:
Hi @ankit-shinde 

Unfortunately no. All RealSense cameras so far **require** USB3, that the Raspberry Pi 3 is missing. 
We are experimenting with adding very **limited** USB2 support to the upcoming D400 camera ([read more](https://github.com/IntelRealSense/librealsense/blob/master/doc/RaspberryPi3.md)), but even for the D400 this feature is experimental (might not make it into the final product), since it comes at severe expense of USB bandwidth, resulting in inferior depth (and no RGB).  

If you are looking for similar form-factor, I would recommend to consider [the Up Board](http://www.up-board.org/upsquared/). People in the community also had limited success running on [NVidia's TX2 boards](http://www.nvidia.com/object/embedded-systems-dev-kits-modules.html). You should be aware that TX2 has known issues in its USB subsystem and is not officially supported by Intel.

closed 2017-12-08 by jlgarcia75

mithunparambath, 2020-03-01:
Dear Dorodnic,
what about Raspberry Pi 4 it has a USB 3. 
What about operating system Raspberry have Linux Debian is it compatible with  Intel real sense D435i and S300.
Best Regards
Mithun