realsenseai/librealsense #877: Multi cameras problem

manhha1402, 2017-12-04:
| Required Info  |   |
|---|---|
| Camera Model | SR300| 
| Firmware Version |   | 
| Operating System & Version | 14.04   |
| Kernel Version (Linux Only)  |   |
Hi,
I tried to connect 2 cameras SR300 but the computer didnt have signals of cameras. The command"lsusb" still detect 2 cameras (Creative Technology ...). I guess that  ports of the computer cannot supply enough power to start streams of cameras. Did anyone have the same problems? Do I need to buy a powered usb 3.0 hub to connect 2 cameras ? 
Thanks

dorodnic, 2017-12-04:
Hi @manhha1402 
Could you please provide the exact `lsusb` and `dmesg` output?
Usually, power related issues show themselves when you try to stream from both devices simultaneously, but they should not prevent enumeration, so there is likely something else at play. 
Are the cameras connected to the same USB3 controller?

manhha1402, 2017-12-05:
Hi @dorodnic 
"lsusb" and "dmesg" after plugging in 1 camera:
[image]
After plugging second camera :
[image]
[image]
Yes, 2 camera are connected to the same USB3 controller

manhha1402, 2017-12-05:
Sorry, the reason is my cable. I tried to use the another cable, is not in the developer kit, so it doesn't work, even it has usb3.0 port. I tested with the cable of sr300, now it works. Thanks

manhha1402, 2017-12-18:
Hi,
I have a question about data collection of multiple cameras. 
When collecting data of a object using cameras, I have problem of Multi-camera synchronization, IR interference as referred here (https://github.com/IntelRealSense/librealsense/issues/138) . 
One solution is to turn off emitters of other cameras while one camera collects data, but librealsense does not support emiters of depth sensor. So I think about turning off depth streams through option(RS2_OPTION_LASER_POWER) but I think it will make data collection time slower than turning off emiters.
Any idea about data collection of multicamera? Thanks

dorodnic, 2017-12-18:
Hi @manhha1402 
Controlling the laser power option is the only way to control SR300 emitter.

closed 2017-12-19 by manhha1402