realsenseai/librealsense #882: Export .ply file from viewer

Panepo, 2017-12-05:
| Required Info  |   |
|---|---|
| Camera Model |  D410 | 
| Firmware Version | 05.06.09.00  | 
| Operating System & Version | Win10 (Build 16299) |
| Kernel Version (Linux Only)  |   |

Hi there,
I am trying to export .ply file from viewer 2.8.2, but the outputed file is unreadable as follows:
```
ply
format binary_little_endian 1.0
comment pointcloud saved from Realsense Viewer
element vertex 691283
property float32 x
property float32 y
property float32 z
property uchar red
property uchar green
property uchar blue
end_header
?0髻曛諫?@灶0??Z7?擢暆k?@S?龍擂輳&0龍擂輳&?0龍擂輳&/擢暆k?@o /擢暆k?@nv-髻曛諫?@??Y蛑? d,?Y蛑??倏i彃P?@j紊i彃P?@砯紊?熂祅	3=紊?熂祅?挪?熂祅.y挪m栽燒b堶禦栽燒b	7倍m栽燒b?玷m栽燒b殦姪m栽燒b浿姪m栽燒b徽矽m栽燒b	?忽?熂祅	魒蕉熂祅	`V阮?熂祅	
=尿i彃P?@	?江i彃P?@	?正[謀"2憭璟謀"{E凶[謀" 疇ˋ[謀" ˋ[謀"SfⅦ[謀"▼[謀" 
?▼i彃P?@ 
..... following is unreadable ....
```
Is there anything wrong with my device or viewer?
Thanks

dorodnic, 2017-12-05:
Hi @Panepo 
The [PLY format](https://en.wikipedia.org/wiki/PLY_(file_format)) can contain data in either textual or binary form. We export data in binary form (that's the **binary_little_endian** in the header) to save space.
You should be able to load this file using [MeshLab](http://www.meshlab.net/) (and perhaps other tools but we verify MeshLab compatibility)
If you want to generate human-readable PLY file, the code should be fairly easy to hack:
[/common/model-views.cpp:199-289](https://github.com/IntelRealSense/librealsense/blob/master/common/model-views.cpp#L199-L289)
You could get rid of:
```cpp
out.close();
out.open(fname, std::ios_base::app | std::ios_base::binary);
```
And instead of `out.write(reinterpret_cast<const char*>(&(new_vertices[i].x)), sizeof(float));` use operator `<<`.

We will add this to the backlog and try to provide both textual and binary formats in one of the future releases.

Panepo, 2017-12-06:
Hi @dorodnic 
For that PLY file, MeshLab(v2016.12) shows a noisy model and Mesh Viewer(v0.3.3) can not open it.
Besides, I am trying to generate PLY file by Python, but there's an error occurs.

```
TypeError                                 Traceback (most recent call last)
<ipython-input-4-d46bfbedc226> in <module>()
     87 
     88         for i in range(0, points.size()):
---> 89             if vertices[i].z:
     90                 print(vertices[i].x)
     91                 print(vertices[i].y)

TypeError: 'pyrealsense2.BufData' object does not support indexing
```
The entire code can be found [here](https://gist.github.com/Panepo/78411f0c50c1581e7bf6319cacc02a35).
Is there any wrong with my code?
Thanks

zivsha, 2018-01-11:
Hi @Panepo ,
Sorry for the late response, hope you've managed to work out this issue (btw: the link you posted is broken).
The vertices from `points.get_vertices()` support the [buffer protocol](https://docs.python.org/3/c-api/buffer.html), you can convert it to a `numpy` array and then use it:
```py
import numpy as np
vertices = np.asanyarray(points.get_vertices())
for i in range(0, points.size()):
    if vertices[i][2]:
        print(vertices[i][0])
        print(vertices[i][1])
```

Panepo, 2018-01-26:
Hi @zivsha 
Thanks this works.
However I found another problem, the python environment ALWAYS crashes if the resolution of the color and depth is not the same. This message pops on windows event viewer:

> Faulting application name: python.exe, version: 3.6.3150.1013, time stamp: 0x59e338b7
> Faulting module name: ucrtbase.dll, version: 10.0.16299.125, time stamp: 0x70f70cc4
> Exception code: 0xc0000409
> Fault offset: 0x000000000006b70e
> Faulting process id: 0x381c
> Faulting application start time: 0x01d3967a103204a0
> Faulting application path: C:\ProgramData\Anaconda3\python.exe
> Faulting module path: C:\WINDOWS\System32\ucrtbase.dll
> Report Id: af310638-a0c9-4112-a5ed-bc9c05ec5a6e

closed 2018-01-26 by Panepo

bingws, 2019-04-28:
> Hi @Panepo
> The [PLY format](https://en.wikipedia.org/wiki/PLY_(file_format)) can contain data in either textual or binary form. We export data in binary form (that's the **binary_little_endian** in the header) to save space.
> You should be able to load this file using [MeshLab](http://www.meshlab.net/) (and perhaps other tools but we verify MeshLab compatibility)
> If you want to generate human-readable PLY file, the code should be fairly easy to hack:
> [/common/model-views.cpp:199-289](https://github.com/IntelRealSense/librealsense/blob/master/common/model-views.cpp#L199-L289)
> You could get rid of:
> 
> ```c++
> out.close();
> out.open(fname, std::ios_base::app | std::ios_base::binary);
> ```
> 
> And instead of `out.write(reinterpret_cast<const char*>(&(new_vertices[i].x)), sizeof(float));` use operator `<<`.
> 
> We will add this to the backlog and try to provide both textual and binary formats in one of the future releases.
 how to generate human-readable PLY file by using python ?

bingws, 2019-04-28:
> Hi @dorodnic
> For that PLY file, MeshLab(v2016.12) shows a noisy model and Mesh Viewer(v0.3.3) can not open it.
> Besides, I am trying to generate PLY file by Python, but there's an error occurs.
>
> ``` [...]

The entire code can't  be found , Can you share? In a hurry

Panepo, 2019-04-29:
Hi @bingws 

[Here](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/export_ply_example.py) is a python export PLY example.

bingws, 2019-04-29:
@Panepo  The demo generates binary files, not ASCII files. I want ASCII readable files.

Panepo, 2019-04-29:
Hi @bingws 

My code is translated from a c++ example to python since there is no python export to ply example at that time. But one year has past, I forget what the code contains and results now.
Sorry

Steven-ZY, 2020-05-26:
@Panepo Excuse me，I run the code you posted about exporting PLY file. I would appreciate it if you can tell me how to only export the x, y, z value of the cloud without the color information. Thanks in advance.