realsenseai/librealsense #885: Segmentation Fault When Extracting Frames

marvin-hao, 2017-12-05:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | 3.15.0.0  | 
| Operating System & Version | Ubuntu 16.04  |
| Kernel Version (Linux Only)  |  4.4.0-101-generic |

I ran 4 SR300 cameras on one computer and utilized color, depth and ir streams at full resolution. After a while, I came across the following error (can replicate but not deterministic):

Thread 1 "python3.6" received signal SIGSEGV, Segmentation fault.
0x00007fffccdc99f1 in librealsense::frame::release (this=0xfd3558) at /home/arb/librealsense/src/archive.cpp:273
273             owner->unpublish_frame(this);

Backtrace:

#0  0x00007fffccdc99f1 in librealsense::frame::release (this=0xfd3558) at /home/arb/librealsense/src/archive.cpp:273
No locals.
#1  0x00007fffccf1fe6e in rs2_release_frame (frame=0xfd3558) at /home/arb/librealsense/src/rs.cpp:813
        __FUNCTION__ = "rs2_release_frame"
        __PRETTY_FUNCTION__ = "void rs2_release_frame(rs2_frame*)"
#2  0x00007fffcd7275ee in rs2::frame::~frame (this=<optimized out>, __in_chrg=<optimized out>) at /usr/local/include/librealsense2/hpp/rs_frame.hpp:172
No locals.
#3  rs2::frameset::foreach<rs2::frameset::first_or_default(rs2_stream) const::{lambda(rs2::frame)#1}>(rs2::frameset::first_or_default(rs2_stream) const::{lambda(rs2::frame)#1}) const (action=..., this=<synthetic pointer>) at /usr/local/include/librealsense2/hpp/rs_frame.hpp:523
        fref = 0xfd3558
        i = 1
        e = 0x0
#4  rs2::frameset::first_or_default (s=RS2_STREAM_COLOR, this=<synthetic pointer>) at /usr/local/include/librealsense2/hpp/rs_frame.hpp:479
No locals.
#5  rs2::frameset::get_color_frame (this=<synthetic pointer>) at /usr/local/include/librealsense2/hpp/rs_frame.hpp:498
        f = {frame_ref = 0x0}

The library is compiled from source. Realsense Viewer works well.
Here is the source code:

for (...){
        rs2::frameset frameset = pipe.wait_for_frames();

        auto color_frame = frameset.get_color_frame();
        auto depth_frame = frameset.get_depth_frame();
        auto ir_frame = frameset.first_or_default(RS2_STREAM_INFRARED);

       ...
    }

marvin-hao, 2017-12-05:
All four cameras are run in separate Python3.6 interpreter sessions/processes.

marvin-hao, 2017-12-06:
Whenever this error happens, it shows _owner_ is a NULL pointer.

(gdb) p owner
$1 = std::shared_ptr (empty) 0x0

I also have a backtrace which starts from the v4l2 backend:

#0  0x00007fffccdc99f1 in librealsense::frame::release (this=0xffa680) at /home/arb/librealsense/src/archive.cpp:273
#1  0x00007fffccf6e739 in librealsense::frame_holder::operator=(librealsense::frame_holder&&) (this=0x7fffbc04ea38, 
    other=<unknown type in /usr/local/lib/librealsense2.so.2, CU 0x9e9be9, DIE 0xa31738>) at /home/arb/librealsense/src/types.cpp:75
#2  0x00007fffccd6d344 in librealsense::pipeline_processing_block::handle_frame (this=0xf49620, frame=..., source=0xf49740)
    at /home/arb/librealsense/src/pipeline.cpp:34
#3  0x00007fffccd6cd90 in librealsense::pipeline_processing_block::pipeline_processing_block(std::vector<int, std::allocator<int> > const&)::{lambda(librealsense::frame_holder, librealsense::synthetic_source_interface*)#1}::operator()(librealsense::frame_holder, librealsense::synthetic_source_interface*) const () at /home/arb/librealsense/src/pipeline.cpp:18
#4  0x00007fffccd77346 in librealsense::internal_frame_processor_callback<librealsense::pipeline_processing_block::pipeline_processing_block(const std::vector<int>&)::<lambda(librealsense::frame_holder, librealsense::synthetic_source_interface*)> >::on_frame(rs2_frame *, rs2_source *) (this=0xe64a10, 
    f=0xf6fc88, source=0xd55e10) at /home/arb/librealsense/src/sync.h:36
#5  0x00007fffccfad527 in librealsense::processing_block::invoke (this=0xf49620, f=...) at /home/arb/librealsense/src/align.cpp:68
#6  0x00007fffccd7032f in librealsense::pipeline::<lambda(librealsense::frame_holder)>::operator()(librealsense::frame_holder) const (
    __closure=0xe0ade8, fref=...) at /home/arb/librealsense/src/pipeline.cpp:494
#7  0x00007fffccd7725e in librealsense::internal_frame_callback<librealsense::pipeline::unsafe_start(std::shared_ptr<librealsense::pipeline_config>)::<lambda(librealsense::frame_holder)> >::on_frame(rs2_frame *) (this=0xe0ade0, fref=0xf6fc88) at /home/arb/librealsense/src/pipeline.cpp:397
#8  0x00007fffccfc9e9b in librealsense::frame_source::invoke_callback (this=0xf49290, frame=...) at /home/arb/librealsense/src/source.cpp:114
#9  0x00007fffccfad66f in librealsense::synthetic_source::frame_ready (this=0xf49350, result=...) at /home/arb/librealsense/src/align.cpp:79
#10 0x00007fffccec4401 in librealsense::syncer_proccess_unit::syncer_proccess_unit()::{lambda(librealsense::frame_holder, librealsense::synthetic_source_interface*)#2}::operator()(librealsense::frame_holder, librealsense::synthetic_source_interface*) const () at /home/arb/librealsense/src/sync.cpp:42
#11 0x00007fffccecd108 in librealsense::internal_frame_processor_callback<librealsense::syncer_proccess_unit::syncer_proccess_unit()::<lambda(librealsense::frame_holder, librealsense::synthetic_source_interface*)> >::on_frame(rs2_frame *, rs2_source *) (this=0xd9b1b0, f=0xfd0e58, source=0xe48130)
    at /home/arb/librealsense/src/sync.h:36
#12 0x00007fffccfad527 in librealsense::processing_block::invoke (this=0xf49230, f=...) at /home/arb/librealsense/src/align.cpp:68
#13 0x00007fffccd7040f in librealsense::pipeline::<lambda(librealsense::frame_holder)>::operator()(librealsense::frame_holder) const (
    __closure=0xda07b8, fref=...) at /home/arb/librealsense/src/pipeline.cpp:506
#14 0x00007fffccd771a2 in librealsense::internal_frame_callback<librealsense::pipeline::unsafe_start(std::shared_ptr<librealsense::pipeline_config>)::<lambda(librealsense::frame_holder)> >::on_frame(rs2_frame *) (this=0xda07b0, fref=0xfd0e58) at /home/arb/librealsense/src/pipeline.cpp:397
#15 0x00007fffccfc9e9b in librealsense::frame_source::invoke_callback (this=0xf11148, frame=...) at /home/arb/librealsense/src/source.cpp:114
#16 0x00007fffcce8b7b6 in librealsense::uvc_sensor::<lambda(librealsense::platform::stream_profile, librealsense::platform::frame_object, std::function<void()>)>::operator()(librealsense::platform::stream_profile, librealsense::platform::frame_object, std::function<void()>) (__closure=0xf4a300, p=..., 
    f=..., continuation=...) at /home/arb/librealsense/src/sensor.cpp:420
#17 0x00007fffcce93afe in std::_Function_handler<void(librealsense::platform::stream_profile, librealsense::platform::frame_object, std::function<void()>), librealsense::uvc_sensor::open(const stream_profiles&)::<lambda(librealsense::platform::stream_profile, librealsense::platform::frame_object, std::function<void()>)> >::_M_invoke(const std::_Any_data &, <unknown type in /usr/local/lib/librealsense2.so.2, CU 0x45e8e0, DIE 0x520ed3>, <unknown type in /usr/local/lib/librealsense2.so.2, CU 0x45e8e0, DIE 0x520ed8>, <unknown type in /usr/local/lib/librealsense2.so.2, CU 0x45e8e0, DIE 0x520edd>) (
    __functor=..., __args#0=<unknown type in /usr/local/lib/librealsense2.so.2, CU 0x45e8e0, DIE 0x520ed3>, 
    __args#1=<unknown type in /usr/local/lib/librealsense2.so.2, CU 0x45e8e0, DIE 0x520ed8>, 
    __args#2=<unknown type in /usr/local/lib/librealsense2.so.2, CU 0x45e8e0, DIE 0x520edd>) at /usr/include/c++/5/functional:1871
#18 0x00007fffccf890fc in std::function<void (librealsense::platform::stream_profile, librealsense::platform::frame_object, std::function<void ()>)>::operator()(librealsense::platform::stream_profile, librealsense::platform::frame_object, std::function<void ()>) const (this=0xdafb20, __args#0=..., 
    __args#1=..., __args#2=...) at /usr/include/c++/5/functional:2267
[... cut ...] :1520
#24 0x00007fffccf848ec in std::thread::_Impl<std::_Bind_simple<librealsense::platform::v4l_uvc_device::stream_on(std::function<void(const librealsense::notification&)>)::<lambda()>()> >::_M_run(void) (this=0xf4a3b0) at /usr/include/c++/5/thread:115
#25 0x00007fffeb6f6c80 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#26 0x00007ffff76926fa in start_thread (arg=0x7fffc8582700) at pthread_create.c:333
#27 0x00007ffff73c8b5d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

zivsha, 2017-12-06:
Hi @marvin-hao ,

Which version are you using?

marvin-hao, 2017-12-06:
@zivsha 

In master branch. The most recent log from my local git repo is:

commit 5fb58db832d3fae75c19ba06689e8de53f5a9add
Merge: df6cd79 d346693
Author: Sergey Dorodnicov <sergey.dorodnicov@intel.com>
Date:   Fri Dec 1 03:32:48 2017 -0800

    Merge pull request #871 from IntelRealSense/python-doc
    
    Updating python documentation

marvin-hao, 2017-12-07:
I added NULL pointer checking for the member variable *owner* of realsense::frame in *src/archive.cpp: void frame::release()* and *src/source.cpp: void frame_source::invoke_callback(frame_holder frame) const*. It works well so far, but I'm not sure if this will cause potential memory leak or further hidden bugs.

zivsha, 2017-12-07:
Can you please share a code snippet / sample that reproduces this issue so we can try it?

marvin-hao, 2017-12-08:
Sorry, most of the confidential code interleaves within the camera control code, but I can share the control flow.
We wrote a Python wrapper of rs2 and we run 4 CPython interpreter sessions on a single computer.

1. Camera initialization:
try {
        self->pipe = rs2::pipeline();
        if (!setup_streams(self))
            return NULL;

        self->pipe_profile = self->pipe.start(self->conf);

        if (!setup_intrinsics(self))
            return NULL;

        self->dev = self->pipe_profile.get_device();
        std::vector<rs2::sensor> sensors = self->dev.query_sensors();

        // For SR300
        if (sensors.size() != 2) {
            PyErr_SetString(PyExc_RuntimeError, "...");
            return NULL;
        }

        if (self->depth_sensor = sensors[0].as<rs2::depth_sensor>()){
            self->depth_scale = self->depth_sensor.get_depth_scale();
        } else {
            PyErr_SetString(PyExc_RuntimeError, "...");
            return NULL;
        }

        // RGB Camera
        if (self -> roi_sensor = sensors[1].as<rs2::roi_sensor>()){}
        else {
            PyErr_SetString(PyExc_RuntimeError, "...");
            return NULL;
        }
    } catch (rs2::error e) {
        PyErr_SetString(PyExc_RuntimeError, e.what());
    }

2. Streaming

In an infinite loop:

for (uint i = 0; i < maxFrames; ++i){
        rs2::frameset frameset = pipe.wait_for_frames();
        try {
            auto color_frame = frameset.get_color_frame();
            auto depth_frame = frameset.get_depth_frame();
            auto ir_frame = frameset.first_or_default(RS2_STREAM_INFRARED);

            // If come across bad frames, decrease the counter by 1 and continue.
            if (
                    !ir_frame || ir_frame.get_data() == NULL ||
                    !color_frame || color_frame.get_data() == NULL ||
                    !depth_frame || depth_frame.get_data() == NULL) {

                --i;
                continue;
            }

            CompFrame memData = {
                    (uint8_t *)(color_frame.get_data()),
                    (uint16_t *)(depth_frame.get_data()),
                    (uint8_t *)(ir_frame.get_data())
            };

            ... (calculation with uses intrinsics and extrinsics)

        } catch (std::exception &e) {
            throw e;
        }
    }

We found after we patched the source files I commented before, there's no more segmentation fault. My concern is that if simply skipping null owner will cause a memory leak or other bugs.

dorodnic, 2017-12-17:
Hi @marvin-hao 
I couldn't find the exact problem in the code, but the symptoms you describe are indicative of frame object being released more then once. While this should not happen when using `rs2::frame` class, if you do use some of the C APIs directly this would be the place to review.

marvin-hao, 2017-12-18:
we do use some C APIs like rs2_deproject_pixel_to_point to align frames. Is there any available C++ API exposed already? We did use C++ API to align in rs1.

dorodnic, 2017-12-19:
There is no limitation to mix C and C++ APIs, however, my concern is that the function `rs2_release_frame` is being called twice on the same frame. 
There is a useful CMake flag `-DTRACE_API=true` that would make such situation very visible in the log.

dorodnic, 2018-01-13:
@marvin-hao please open-new issue if you reproduce this problem (closing for now)

closed 2018-01-13 by dorodnic

KeCh96, 2018-05-31:
I also run into `Segmentation fault` error. Did you solve this error by patching `src/archive.cpp` and `src/source.cpp`? @marvin-hao 

Could you please show your solution? Although you have shared your `Camera initialization` code and `Streaming` above, I still confused about how to edit `src/archive.cpp` and `src/source.cpp`.

I am puzzled by this problem for several days, my detailed situation is [here](https://github.com/IntelRealSense/librealsense/issues/1785). Could you provide your solution? Thanks in advance.