realsenseai/librealsense #886: Converting (u,v) to (x,y,z) for aligned image and depth

danielw256, 2017-12-06:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | 3.21.0.0 | 
| Operating System & Version | Win10 Pro version 1709 build 16299.19 |

Is this the correct process for converting (u,v) to (x,y,z) with respect to the image camera origin for an aligned image and depth frame?
rs2_deproject_pixel_to_point(rgb_points, &intrin_rgb, rgb_pixels, rgb_aligned_distance);

When I continue the process to verify that I can find the original distance pixel, my z with respect to depth sensor seems to be off by about 5 millimeters?
rs2_transform_point_to_point(depth_points, &extrin_rgb_to_depth, rgb_points);
rs2_project_point_to_pixel(depth_pixels, &intrin_depth, depth_points);

(u,v) image = (x,y,z) respect to image sens = (x,y,z) respect to depth sens = (s,t) depth
  (357,196) = (0.093132,-0.099292,0.943774) = (0.117853,-0.095003,0.948431) - (738.910522,279.004242)

(s,t) depth = (x,y,z) respect to depth sens = (x,y,z) respect to depth sens = (u,v) image
  (738,278) = (0.116342,-0.095565,0.943774) = (0.091621,-0.099833,0.939115) = (356.586517,195.425629)

zivsha, 2017-12-06:
Hi @danielw256 ,
The SDK provides a utility for aligning images with respect to depth data as you can see in the [`rs-align` example](https://github.com/IntelRealSense/librealsense/tree/development/examples/align).

The transformation you are asking about can be viewed in the source file of `align.cpp`:
https://github.com/IntelRealSense/librealsense/blob/5871681b538bfda61d337bb6083e9d8b9674d523/src/align.cpp#L484-L531

danielw256, 2017-12-06:
@zivsha 

I've already gone through that alignment, so I have a 1280x720 rgb image and an aligned 1280x720 depth image. 

When my program identifies an important (u,v) pixel in the rgb image, it needs to convert find that pixel into 3D space, as a (x,y,z) point with respect to the camera sensor. I've used:

rs2_deproject_pixel_to_point(rgb_points, &intrin_rgb, rgb_pixels, rgb_aligned_distance);

But I've tried to verify the correctness of the transforms to go back all the way to the original depth pixel.
The depth with respect to the depth sensor seems to be off by about 5mm. As these are mathematical transforms, I would expect the error to be a lot less.

Panepo, 2017-12-07:
Hi @zivsha 
How to run these two functions rs2_project_point_to_pixel and rs2_deproject_pixel_to_point in python? Is there an example?
Thanks

jlgarcia75, 2017-12-08:
@danielw256,

What is the actual distance from your object to the depth sensor?

Jesus G.
Intel Customer Support

danielw256, 2017-12-08:
The actual distance is about 0.95m.

It shouldn't matter in verifying the math of the transforms.
Processing block align converts (1) depth pixels w/ 16-bit depth value to depth points (2) to rgb points (3) to rgb pixels w/ 16-bit depth value
I'm trying to convert the rgb pixels to rgb points. I continue the process to find the depth points and original depth pixels to verify I'm doing it correctly.
I'm going from (1) rgb pixels w/ 16-bit depth value to rgb points (2) to depth points (3) to depth pixels w/ 16-bit depth value.

All of my verification Z values for rgb points and depth points seems to be off by about 5mm from the original Z values.

RealSense-Customer-Engineering, 2018-02-28:
[Realsense Customer Engineering Team Comment] 
Hello @danielw256,

You may want to refer to this article: https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0 for  more info on projection. Let us know if this is helpful.

Regards,
Jesus G.
Intel Customer Support

danielw256, 2018-02-28:
Thanks. Closing this issue

closed 2018-02-28 by danielw256

ywjia, 2018-10-19:
@danielw256 what is the conclusion, then, about the 5mm discrepancy of Z values, please?

danielw256, 2018-10-22:
Unfortunately, there is no official answer from Intel. I would guess there is some physical discrepancy between the two camera sensors and some mathematical error