realsenseai/librealsense #889: Depth Camera works but RGB Camera does not work

sktbkvygr, 2017-12-07:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version |  3.21.0.0 | 
| Operating System & Version | Windows 10 Pro Version 1703   |
| Kernel Version (Linux Only)  |  |
| Driver Version (SR300 RGB) | 2006/06/21 10.0.15063.447 |
| Driver Version (SR300 Depth) | 2016/09/22 3.3.27.5718  |
| Driver Version (SR300 Virtual) | 2016/06/08 12.11.38.553  |
| MSVS Version | VS2015 Professional |

When realsense-viewer.exe (builded by MSVS ) run and turn on RGB Camera, the following message appears. Depth Sensor works without problems.
Is there anything wrong?
[image]

dorodnic, 2017-12-17:
Hi @sktbkvygr 

We have not encountered this problem before (the error message is somewhat misleading, because it mistakes whatever is going on with known Linux issue, where kernel patch can solve the problem)

If I understand correctly, both SR300 Depth and SR300 Color imaging devices are enumerated in the Device Manager, but when you start streaming Color frames never show up, is this correct? 

If you have the Virtual Driver showing up under Imaging Devices, could you please disable the Intel Digital Camera Manager service? I would also try uninstalling camera drivers, and reconnecting the camera.

sktbkvygr, 2017-12-25:
Hi @dorodnic 

As a result the cause was a hardware failure.
A replacement SR300 is working normally.
Thanks.

closed 2017-12-25 by sktbkvygr