realsenseai/librealsense #890: (sync.cpp:221) Stream matcher not found! stream=Depth

tchapi, 2017-12-07:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | 3.21.0.0  | 
| Operating System & Version | Ubuntu 16.04.2 LTS  |
| Kernel Version (Linux Only)  | 4.4.0-101-generic  |

Hello,

I have a weird error when scanning : 

    ERROR [140502095603456] (sync.cpp:221) Stream matcher not found! stream=Depth

Here is the process I do (C++11) : 

```
rs2::context ctx;
rs2::device dev;
rs2::pipeline pipe;
rs2::pointcloud pc;
rs2::points points;

rs2::log_to_console(RS2_LOG_SEVERITY_ERROR);

rs2::pipeline_profile profile = pipe.start();

auto sensor = profile.get_device().first<rs2::depth_sensor>();
scale = sensor.get_depth_scale();

for (auto i = 0; i < 30; ++i) pipe.wait_for_frames();

rs2::frameset frames = pipe.wait_for_frames();

rs2::depth_frame depth_frame = frames.get_depth_frame();
rs2::frame color_frame = frames.get_color_frame();

points = pc.calculate(depth_frame);
pc.map_to(color_frame);
```

The error seems to be outputted when I do the loop waiting for frames. Generally, the process continues without problem even with the error but sometimes, I get a frame that has only 0-depth points at the end;

Any idea ? Am I missing something ?
Thanks a lot

aangerma, 2017-12-13:
We are working to resolve the first issue.
Regarding the zero depth can you please check if you have depth by running the realsense-viewer? 
and can you add more details: do you get zero points or all the points are zero? and what do you mean "at the end"?

tchapi, 2017-12-13:
Hi @aangerma and thanks a lot for your answer.

After more testing today, I have a feeling the error message (`Stream matcher not found! stream=Depth`) is not linked to the points being zero-depth (I think it's the cable that is weak, or the USB3 port is not sufficiently powered sometimes).

The **realsense-viewer** seems to work fine, and I have tested on another machine (supposedly with better USB3 management) and even though the error message appears, the points seem to be ok.

dorodnic, 2017-12-17:
Hi @tchapi 
`stream matcher not found` problem should be resolved on the development branch.
Do you need any further assistance to help troubleshoot the zero-depth behavior?

tchapi, 2017-12-18:
Hi @dorodnic 
Thanks ! I'll have a look ASAP. As for the zero-depth, I'll investigate myself first and create an issue if I can have a _reproducable_ example

Best

closed 2017-12-18 by tchapi