realsenseai/librealsense #891: Multiple pipes for the same device

sogartar, 2017-12-07:
| Required Info  |   |
|---|---|
| Camera Model | D400 | 
| Firmware Version |   | 
| Operating System & Version |   |
| Kernel Version (Linux Only)  |   |

I don't have the sensor yet so I can not check if this is the case. My question is could I have multiple pipes for the same device? In the capture example depth, color and ir frames could be captured from the same pipe with wait_for_frames(). Could you have each stream coming from its own pipe? If this is possible, would stream profiles from different pipes work as expected? For example would stream_profile::get_extrinsics_to behave OK?

sogartar, 2017-12-07:
After browsing through the source a little more, I have more clarity.

Create a `pipeline` without starting it. Cast it to `std::shared_ptr<rs2_pipeline>` and `resolve()` it with a `config` to get a `pipeline_profile`. From it get `stream_profile`s.
From the `context` I can get all sensors. For each `sensor` I can `get_stream_profiles()`, that can be provided by the `sensor`. Match the one from my configured stream profiles by `rs2_stream` type. Оpen the `sensor` with it. Then `start` the sensor with my callback.
I can see one other option to do the configuration. Not to use `config`, but to use `advanced_mode` and then get stream profiles from the `device`.

zivsha, 2017-12-07:
Hi @sogartar ,

The SDK provides 2 main APIs for users:
1. `rs2:context` + `rs2::device` + `rs2::sensor`
2. `rs2::pipeline`

The second API is a facade (simpler API) over the first one.
It is not recommended to mix to two together but for advanced stuff using the pipeline is just not enough and you'll have to use the first set of APIs.

Regarding your first question about multiple pipes:
First, let's have some common terminology: 
* A device is a collection of sensors. (e.g: D400)
* A sensor is a collection of streams (e.g: Depth sensor which is comprised of IR+DEPTH)
* A stream represents the flow of a single data type (e.g: IR / Depth / Color stream)
Read more on this in the documentation of the [`rs-sensor-control` example](https://github.com/IntelRealSense/librealsense/tree/master/examples/sensor-control).

The pipeline is (at the moment) a wrapper of a device for simplified usage which removes the boilerplate code required to start streaming. A single pipeline uses a single device (once it is started - after you stop it the pipeline can use a different device if configured differently).

Technically I don't think we prevent creating multiple pipeline which use different streams of the same device, the only restriction is that "start" (of streams) is at the sensor level - meaning that all streams of the same sensor must be started together (via the sensor's start function) and stopping the sensor will stop all streams. So if you have 2 pipelines , 1 D400 camera and you plan to start IR from 1 pipeline, and DEPTH form another pipeline - you will fail (because the D400 has a single sensor that has DEPTH and IR streams).

From your first question it looks like you want to have more control over each sensor (e.g get IR+DEPTH frame on 1 thread and COLOR frames on another thread), if the is the case don't use the pipeline and follow the `rs-sensor-control` example.

Regarding your second comment:
The `rs2::config` serves as a configuration parameter for the `rs2::pipeline`. You can think of it as a collection of your requests / required restriction that you want to have the pipeline to make.
The usage should be:
* create a `rs2::pipeline`
* create a `rs2::config`
* configure the config
* start the pipeline with the config
See examples in our wiki's [API-How-To](https://github.com/IntelRealSense/librealsense/wiki/API-How-To#start-streaming-left-and-right-infrared-imagers) and in PR #672.

FYI: You don't need to explicitly cast the pipeline to `std::shared_ptr<rs2_pipeline>` since it can be implicitly casted.

Please try to follow all the above documentation and examples, and if this doesn't answer your question, maybe explain what it is that your are trying to accomplish and we will try to suggest the way

dorodnic, 2017-12-24:
Hi @sogartar 
Please let us know if you have additional questions on this topic

closed 2018-01-13 by dorodnic