realsenseai/librealsense #892: rs-capture Frame didn't arrived within 5000

fouvy, 2017-12-07:
SR300 doesn't work with command rs-capture
RealSense error calling rs2_pipeline_wait_for_frames(pipe:0xd88b20):
    Frame didn't arrived within 5000
**cmd:**
`sudo dmesg | tail -n 50`
**result:**
[  548.205255] R10: 00007fc591131be0 R11: 0000000000000202 R12: 0000000000000009
[  548.205256] R13: 00007fc591131be0 R14: ffffffffffffff92 R15: 0000000002347d00
[  548.205258] ---[ end trace fa8ad5ba2d680811 ]---
[  548.205339] ------------[ cut here ]------------
[  548.205346] WARNING: CPU: 3 PID: 9396 at /build/linux-hwe-zOpU13/linux-hwe-4.10.0/fs/sysfs/group.c:237 sysfs_remove_group+0x8c/0x90
[  548.205347] sysfs group 'power' not found for kobject 'media0'
[  548.205348] Modules linked in: hid_sensor_gyro_3d(OE) hid_sensor_accel_3d(OE) hid_sensor_trigger industrialio_triggered_buffer kfifo_buf hid_sensor_iio_common hid_sensor_hub industrialio uvcvideo(OE) videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_core videodev(OE) media snd_usb_audio snd_usbmidi_lib ccm rfcomm pci_stub vboxpci(OE) vboxnetadp(OE) vboxnetflt(OE) vboxdrv(OE) bbswitch(OE) bnep nls_iso8859_1 rt2800usb intel_rapl rt2x00usb rt2800lib x86_pkg_temp_thermal intel_powerclamp rt2x00lib coretemp mxm_wmi snd_hda_codec_hdmi snd_hda_codec_realtek snd_hda_codec_generic arc4 kvm_intel snd_hda_intel kvm snd_hda_codec snd_hda_core snd_hwdep irqbypass iwlmvm snd_pcm snd_seq_midi snd_seq_midi_event crct10dif_pclmul mac80211 input_leds serio_raw crc32_pclmul btusb btrtl ghash_clmulni_intel
[  548.205394]  snd_rawmidi nvidia_uvm(POE) pcbc snd_seq snd_seq_device snd_timer iwlwifi rtsx_pci_ms aesni_intel memstick snd aes_x86_64 crypto_simd cfg80211 glue_helper cryptd soundcore mei_me intel_cstate shpchp mei intel_pch_thermal intel_rapl_perf hci_uart btbcm btqca btintel bluetooth wmi intel_lpss_acpi intel_lpss tpm_crb acpi_pad joydev mac_hid ppdev lp parport autofs4 hid_generic usbhid nvidia_drm(POE) i915 nvidia_modeset(POE) rtsx_pci_sdmmc nvidia(POE) i2c_algo_bit drm_kms_helper psmouse syscopyarea sysfillrect sysimgblt fb_sys_fops nvme r8169 drm ahci nvme_core rtsx_pci libahci mii pinctrl_sunrisepoint i2c_hid video pinctrl_intel hid fjes [last unloaded: videobuf2_memops]
[  548.205446] CPU: 3 PID: 9396 Comm: rs-capture Tainted: P        W  OE   4.10.0-40-generic #44~16.04.1-Ubuntu
[  548.205447] Hardware name: Notebook                         P65_P67RGRERA/P65_P67RGRERA, BIOS 1.05.13 01/27/2016
[  548.205448] Call Trace:
[  548.205451]  dump_stack+0x63/0x90
[  548.205454]  __warn+0xcb/0xf0
[  548.205457]  warn_slowpath_fmt+0x5f/0x80
[  548.205461]  ? kernfs_find_and_get_ns+0x48/0x60
[  548.205466]  sysfs_remove_group+0x8c/0x90
[  548.205469]  dpm_sysfs_remove+0x57/0x60
[  548.205473]  device_del+0x12d/0x350
[  548.205480]  media_devnode_unregister+0x3f/0x60 [media]
[  548.205485]  media_device_unregister+0x147/0x180 [media]
[  548.205490]  uvc_delete+0x6d/0x160 [uvcvideo]
[  548.205494]  uvc_release+0x23/0x30 [uvcvideo]
[  548.205503]  v4l2_device_release+0xe6/0x120 [videodev]
[  548.205506]  device_release+0x32/0x90
[  548.205510]  kobject_release+0x6a/0x170
[  548.205513]  kobject_put+0x27/0x50
[  548.205516]  put_device+0x17/0x20
[  548.205524]  v4l2_release+0x50/0x80 [videodev]
[  548.205527]  __fput+0xe7/0x220
[  548.205531]  ____fput+0xe/0x10
[  548.205535]  task_work_run+0x7e/0xa0
[  548.205539]  do_exit+0x2c8/0xb40
[  548.206090]  ? _nv006446rm+0x17e/0x1f0 [nvidia]
[  548.206095]  do_group_exit+0x43/0xb0
[  548.206098]  get_signal+0x293/0x620
[  548.206103]  do_signal+0x37/0x750
[  548.206106]  ? do_futex+0x214/0x540
[  548.206111]  ? mntput+0x24/0x40
[  548.206115]  ? __check_object_size+0x108/0x1e3
[  548.206120]  exit_to_usermode_loop+0x76/0xb0
[  548.206123]  syscall_return_slowpath+0x59/0x60
[  548.206128]  entry_SYSCALL_64_fastpath+0xab/0xad
[  548.206130] RIP: 0033:0x7fc5954a4709
[  548.206131] RSP: 002b:00007fc591131b10 EFLAGS: 00000202 ORIG_RAX: 00000000000000ca
[  548.206134] RAX: fffffffffffffdfc RBX: 00007fc591131c70 RCX: 00007fc5954a4709
[  548.206136] RDX: 0000000000000009 RSI: 0000000000000189 RDI: 0000000002348b64
[  548.206137] RBP: 00007fc591131b80 R08: 0000000002348b38 R09: 00000000ffffffff
[  548.206139] R10: 00007fc591131be0 R11: 0000000000000202 R12: 0000000000000009
[  548.206140] R13: 00007fc591131be0 R14: ffffffffffffff92 R15: 0000000002347d00
[  548.206143] ---[ end trace fa8ad5ba2d680812 ]---
==============================================================================
| Required Info  |   |
|---|---|
| Camera Model |  SR300 | 
| Firmware Version |  2.8.2 | 
| Operating System & Version |   ubuntu 16.04 |
| Kernel Version (Linux Only)  |  4.10.0-40-generic  |

fgzhu, 2017-12-12:
i have encounter the same problem on surface laptop and surface pro with win 10.
if connect the sensor to a usbhub and then connect to the laptop, the sensor works fine.

fouvy, 2017-12-13:
I have solved this problem. My SR300 depth camera might have problem. I gave it to retailer, and he gave me a new SR300 module. 
I found an interesting problem: ubuntu 16.04 must update and dist-upgrade to new version

`sudo apt update && sudo apt dist-upgrade`

After upgraded ubuntu 16.04 kernel and software, when you install librealsense, you must PULL OUT your SR300 connection, and run this code:

```
dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge
sudo apt-get install realsense-uvcvideo
sudo apt-get install librealsense2-utils
```

Driver will demod successfully.

It might be some patch problem of installing kernel mod when your connect SR300 with your computer.

fouvy, 2017-12-13:
I also solved this problem in windows 10 platform.

It's very interesting that intel driver doesn't run OK in windows 10. SR300 depth camera is running microsoft's driver. when you connect your SR300 with windows, you will find that windows has identified your device. After your install your Firmware driver, windows is still running microsoft's driver. OK, double click your Firmware driver, and UNINSTALL your driver, this time, SR300 driver will change into intel's driver. then SR300 would work OK. The solving steps is:
1. connect device to windows 10
2. install intel firmware driver
3. double click firmware driver and remove firmware driver
After this steps, SR300 will run intel's driver not microsofte's driver.
It's very interesting....
ps. my fireware is:  intel_rs_dcm_sr300_3.1.25.2599.exe
It came from my retailer.

dorodnic, 2017-12-13:
Hi @fouvy 

Once Linux kernel 4.12 becomes more common it will simplify the process a bit for Linux users, since no driver installation will be required. You still have to re-connect the camera to re-apply the udev-rules, I believe. 

On Windows, we are gradually trying to deprecate the DCM driver. For some time now librealsense doesn't require it to operate and works fine with Microsoft UVC driver. Usually, letting Windows Update to its thing and then uninstalling the driver to go back to the standard UVC device works.

closed 2017-12-13 by dorodnic

elfslmn, 2018-06-01:
Hi @fouvy 
I have same error in Ubuntu 16.04 with SR300. I try your solution there is no "realsense" entry in dpkg. Do you know what can be reason of that?