realsenseai/librealsense #894: kernel update issue on ubuntu 14.04.5

LZhipeng, 2017-12-07:
| Required Info  |   |
|---|---|
| Camera Model | R200 / F200 / SR300 / ZR300 / D400 | 
| Firmware Version |   | 
| Operating System & Version |   |
| Kernel Version (Linux Only)  |   |
| SDK Version | |

I follow the https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md guide to update the linux kernel. There error after I run the update-grub:
root@ubuntu:~# update-grub
/usr/sbin/grub-probe: error: failed to get canonical path of `/cow'.

How to resolve this error?

Thanks!

dorodnic, 2017-12-11:
This is rather general question about [grub](https://askubuntu.com/questions/254491/failed-to-get-canonical-path-of-cow), not really related to librealsense. 
Are you running from live-CD?

LZhipeng, 2017-12-12:
No this issue in another installation.
But another issue appeared. 
Pls check https://github.com/IntelRealSense/librealsense/issues/898

Thanks!

closed 2017-12-21 by LZhipeng