realsenseai/librealsense #895: linking to XConvertSelection

alexge233, 2017-12-08:
I'm trying to build librealsense on a Tegra TX2, and unless I turn off `-DBUILD_GRAPHICAL_EXAMPLES=Off` during cmake, I get link errors to `XConvertSelection` and other X11 methods. I am guessing that cmake does not search for the package/libraries?

dorodnic, 2017-12-13:
The [realsense-viewer](https://github.com/IntelRealSense/librealsense/blob/master/tools/realsense-viewer/CMakeLists.txt#L25) is searching for GTK that might be related. However, we do compile the SDK on the TX2 using the following [guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md) and don't see this problem. 
Please update the community if you find a solution.

alexge233, 2017-12-13:
@dorodnic Ok, that makes sense, since I've turned it off it works fine.
I'm checking out tag v1.12.1 and testing it right now on a TX2, so if anything arises I'll comment back.
I guess the right approach would be check if the library/package is present in CMakeLists.txt.

closed 2017-12-13 by alexge233