realsenseai/librealsense #896: 'No Frame Received' error in realsense-viewer after a source code update

marvin-hao, 2017-12-08:
| Required Info  |   |
|---|---|
| Camera Model |  SR300 | 
| Firmware Version |  3.10.10.0  | 
| Operating System & Version |  Ubuntu 16.04 |
| Kernel Version (Linux Only)  | 4.10.0-42-generic  |
| SDK Version | rs2 |

Working commit: 4ed892bf4037102844e2dc361af493ca8ec92080

Commit not working: 0cc659d985af42f311b2c1c6aa9664d17133621d

dorodnic, 2017-12-17:
Hi @marvin-hao 
Could you please provide more information? We have noticed some performance degradation between the two commits you mentioned, and since then submitted a fix to the development branch.
Could you let us know if it helped?

marvin-hao, 2017-12-18:
Can you please help me locate the commit in the dev branch that fixes the issue?

dorodnic, 2017-12-19:
This is the Pull-Request I was referring to: #907

dorodnic, 2018-01-13:
@marvin-hao did the camera worked in the Viewer with the latest release (2.9.0)?

closed 2018-01-18 by dorodnic