realsenseai/librealsense #898: Errors when run examples in librealsense

lzppzlhotmail, 2017-12-11:
| Required Info  |   |
|---|---|
| Camera Model | R200 / F200 / SR300 / ZR300 / D400 | 
| Firmware Version |   | 
| Operating System & Version | Ubuntu 14.04.5 |
| Kernel Version (Linux Only)  | 4.4.0-103  |
| SDK Version | 2.0 |

I have followed the steps in https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md and finished successfully.
root@lizp-ECS-1830V2NA:~/librealsense/doc# dmesg |grep -n 50
71:[    0.000000] reg 7, base: 17150MB, range: 2MB, type UC
170:[    0.000000] *BAD*gran_size: 8M   chunk_size: 512M        num_reg: 10    lose cover RAM: -250M
222:[    0.000000] BRK [0x02215000, 0x02215fff] PGTABLE
241:[    0.000000] NODE_DATA(0) allocated [mem 0x42fdf5000-0x42fdf9fff]
328:[    0.000000] tsc: Detected 3093.050 MHz processor
331:[    0.000025] ACPI: Core revision 20150930
362:[    0.022250] DMAR: Host address width 39
402:[    0.094559] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
407:[    0.103507] cpuidle: using governor ladder
408:[    0.119508] cpuidle: using governor menu
471:[    0.149150] pci 0000:00:1c.0: [8086:8c10] type 01 class 0x060400
496:[    0.149985] pci 0000:00:1f.3: [8086:8c22] type 00 class 0x0c0500
497:[    0.149998] pci 0000:00:1f.3: reg 0x10: [mem 0xf7a15000-0xf7a150ff 64bit]
498:[    0.150014] pci 0000:00:1f.3: reg 0x20: [io  0xf040-0xf05f]
499:[    0.150089] pci 0000:01:00.0: [10de:1401] type 00 class 0x030000
500:[    0.150101] pci 0000:01:00.0: reg 0x10: [mem 0xf6000000-0xf6ffffff]
501:[    0.150106] pci 0000:01:00.0: reg 0x14: [mem 0xe0000000-0xefffffff 64bit pref]
502:[    0.150112] pci 0000:01:00.0: reg 0x1c: [mem 0xf0000000-0xf1ffffff 64bit pref]
503:[    0.150116] pci 0000:01:00.0: reg 0x24: [io  0xe000-0xe07f]
504:[    0.150119] pci 0000:01:00.0: reg 0x30: [mem 0xf7000000-0xf707ffff pref]
505:[    0.150155] pci 0000:01:00.0: System wakeup disabled by ACPI
506:[    0.150177] pci 0000:01:00.1: [10de:0fba] type 00 class 0x040300
507:[    0.150189] pci 0000:01:00.1: reg 0x10: [mem 0xf7080000-0xf7083fff]
508:[    0.156650] pci 0000:00:01.0: PCI bridge to [bus 01]
595:[    0.182377] pnp 00:05: Plug and Play ACPI device, IDs PNP0501 (active)
597:[    0.182503] pnp 00:06: Plug and Play ACPI device, IDs PNP0501 (active)
599:[    0.182625] pnp 00:07: Plug and Play ACPI device, IDs PNP0501 (active)
601:[    0.182746] pnp 00:08: Plug and Play ACPI device, IDs PNP0501 (active)
609:[    0.183189] pnp 00:0d: Plug and Play ACPI device, IDs PNP0501 (active)
611:[    0.183317] pnp 00:0e: Plug and Play ACPI device, IDs PNP0501 (active)
613:[    0.183504] pnp 00:0f: Plug and Play ACPI device, IDs PNP0400 (active)
623:[    0.183789] system 00:11: [mem 0xfed45000-0xfed8ffff] has been reserved
632:[    0.189500] pci 0000:00:1c.0: bridge window [mem 0x00100000-0x000fffff 64bit pref] to [bus 02] add_size 200000 add_align 100000
633:[    0.189501] pci 0000:00:1c.0: bridge window [mem 0x00100000-0x000fffff] to [bus 02] add_size 200000 add_align 100000
684:[    0.190050] UDP-Lite hash table entries: 8192 (order: 6, 262144 bytes)
711:[    0.440716] Asymmetric key parser 'x509' registered
729:[    0.441503] input: Power Button as /devices/LNXSYSTM:00/LNXPWRBN:00/input/input1
730:[    0.441504] ACPI: Power Button [PWRF]
732:[    0.441950] ACPI: Thermal Zone [TZ00] (28 C)
736:[    0.442194] Serial: 8250/16550 driver, 32 ports, IRQ sharing enabled
737:[    0.462545] 00:05: ttyS4 at I/O 0x210 (irq = 10, base_baud = 115200) is a 16550A
738:[    0.482955] 00:06: ttyS5 at I/O 0x218 (irq = 10, base_baud = 115200) is a 16550A
739:[    0.503338] 00:07: ttyS6 at I/O 0x220 (irq = 10, base_baud = 115200) is a 16550A
740:[    0.523740] 00:08: ttyS7 at I/O 0x228 (irq = 10, base_baud = 115200) is a 16550A
741:[    0.544138] 00:0d: ttyS0 at I/O 0x3f8 (irq = 4, base_baud = 115200) is a 16550A
742:[    0.564538] 00:0e: ttyS1 at I/O 0x2f8 (irq = 3, base_baud = 115200) is a 16550A
778:[    0.598250] ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
818:[    0.605033] microcode: CPU0 sig=0x306c3, pf=0x2, revision=0x17
819:[    0.605049] microcode: CPU1 sig=0x306c3, pf=0x2, revision=0x17
820:[    0.605065] microcode: CPU2 sig=0x306c3, pf=0x2, revision=0x17
821:[    0.605083] microcode: CPU3 sig=0x306c3, pf=0x2, revision=0x17
822:[    0.605090] microcode: CPU4 sig=0x306c3, pf=0x2, revision=0x17
828:[    0.605495] Loading compiled-in X.509 certificates
829:[    0.606079] Loaded X.509 cert 'Build time autogenerated kernel key: 7294d9fa1c48c2283262986ae1345e8884ee1285'
861:[    0.655082] scsi host4: ahci
936:[    9.080500] ppdev: user-space parallel port driver
961:[   11.415003] snd_hda_intel 0000:01:00.1: Handle vga_switcheroo audio client
1018:[   12.185013] nouveau 0000:01:00.0: DRM: unknown connector type 70
1019:[   12.185035] nouveau 0000:01:00.0: DRM: failed to create encoder 1/8/0: -19
1020:[   12.185036] nouveau 0000:01:00.0: DRM: Unknown-1 has no encoders, removing
1021:[   12.185056] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
1022:[   12.185057] [drm] Driver supports precise vblank timestamp query.
1070:[   22.285070] igb 0000:04:00.0 eth1: igb: eth1 NIC Link is Up 100 Mbps Full Duplex, Flow Control: RX/TX
1084:[ 2150.705620] usb 3-2.2: new full-speed USB device number 4 using xhci_hcd
1085:[ 2150.950250] usb 3-2.2: New USB device found, idVendor=041e, idProduct=4099
1086:[ 2150.950253] usb 3-2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
1087:[ 2150.950254] usb 3-2.2: Product: Creative VF0800
1088:[ 2150.950255] usb 3-2.2: Manufacturer: Creative Technology Ltd
1089:[ 2150.950256] usb 3-2.2: SerialNumber: 000000032730
1104:[212539.036903] audit: type=1400 audit(1512950149.943:40): apparmor="STATUS" operation="profile_replace" profile="unconfined" name="/usr/lib/cups/backend/cups-pdf" pid=20698 comm="apparmor_parser"
1105:[212539.036908] audit: type=1400 audit(1512950149.943:41): apparmor="STATUS" operation="profile_replace" profile="unconfined" name="/usr/sbin/cupsd" pid=20698 comm="apparmor_parser"
1130:[221073.749141] uvcvideo: Unknown video format 52494150-8536-4841-b6bf-8fc6ffb083a8
1133:[221073.749850] input: Intel(R) RealSense(TM) Camera S as /devices/pci0000:00/0000:00:14.0/usb4/4-2/4-2.1/4-2.1:1.2/input/input21
1164:[221560.305088] usb 3-2.2: new full-speed USB device number 8 using xhci_hcd
1170:[221560.685077] input: Creative Technology Ltd Creative VF0800 as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2.2/3-2.2:1.3/0003:041E:4099.0004/input/input23
1172:[221950.448736] usb 3-2: USB disconnect, device number 7
1173:[221950.448739] usb 3-2.2: USB disconnect, device number 8
1196:[224127.347509] hub 3-2:1.0: USB hub found
1201:[224128.416950] usb 3-2.2: Product: Creative VF0800

There are errors when I run the examples:
root@lizp-ECS-1830V2NA:~/librealsense# ./build/examples/capture/rs-capture
rs-capture: /root/glfw-3.2.1/src/window.c:774: glfwSetWindowUserPointer: Assertion `window != ((void *)0)' failed.
Aborted (core dumped)
root@lizp-ECS-1830V2NA:~/librealsense# ./build/examples/save-to-disk/rs-save-to-disk
RealSense error calling rs2_pipeline_start(pipe:0xeea250):
    No device connected
root@lizp-ECS-1830V2NA:~/librealsense#

What's wrong with the system and how can I resolve these errors?
Thanks!

ev-mp, 2017-12-11:
@lzppzlhotmail ,
It seem that the device does not complete the enumeration correctly either due to power or other USB-related issues.

I got a similar setup with Creative VF0800(SR300) device running on Ubuntu 14.04/4.4.0-103 and the kernel log looks different - note SR300 is recognized as USB3 device when enumerated properly:

[Mon Dec 11 11:52:07 2017] usb 3-7.2: New USB device found, idVendor=041e, idProduct=4099
[Mon Dec 11 11:52:07 2017] usb 3-7.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Mon Dec 11 11:52:07 2017] usb 3-7.2: Product: Creative VF0800
[Mon Dec 11 11:52:07 2017] usb 3-7.2: Manufacturer: Creative Technology Ltd
[Mon Dec 11 11:52:07 2017] usb 3-7.2: SerialNumber: 000000020454
[Mon Dec 11 11:52:07 2017] input: Creative Technology Ltd Creative VF0800 as /devices/pci0000:00/0000:00:14.0/usb3/3-7/3-7.2/3-7.2:1.3/0003:041E:4099.000B/input/input41
[Mon Dec 11 11:52:07 2017] hid-generic 0003:041E:4099.000B: input,hiddev0,hidraw0: USB HID v1.11 Device [Creative Technology Ltd Creative VF0800] on usb-0000:00:14.0-7.2/input3
**[Mon Dec 11 11:52:08 2017] usb 4-5.1: new SuperSpeed USB device number 21 using xhci_hcd
[Mon Dec 11 11:52:08 2017] usb 4-5.1: New USB device found, idVendor=8086, idProduct=0aa5**
[Mon Dec 11 11:52:08 2017] usb 4-5.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Mon Dec 11 11:52:08 2017] usb 4-5.1: Product: Intel(R) RealSense(TM) Camera SR300
[Mon Dec 11 11:52:08 2017] usb 4-5.1: Manufacturer: Intel(R) RealSense(TM) Camera SR300

You should try other USB3 ports or connect it via self-powered USB3 hub.

LZhipeng, 2017-12-12:
I have tried on other ports but not effective.

LZhipeng, 2017-12-18:
Any more comments for this issue? Thanks a lot!

LZhipeng, 2017-12-21:
There is the same issue on Ubuntu 16.04.3.

LZhipeng, 2017-12-21:
@ev-mp Could you give some detail info about your test environment? 
My camera is the Creative FV0800 and the USB cable is about 120cm.
Now the Linux kernel don't think the USB port be the USB3.0.
I have tried the powered USB3.0 Hub but not effective.

LZhipeng, 2017-12-22:
@ev-mp You are right. My camera wasn't found as a USB3.0 device in my system. The reason for this is the Creative FV0800 isn't stable currently. Sometimes it can be found as USB3.0 device.

lzppzlhotmail, 2017-12-25:
I have confirmed that the Creative NF0800 isn't stable because of the USB3.0 port problem.

closed 2017-12-25 by lzppzlhotmail

bf826989019, 2019-01-19:
I have this issue too. How can I do ?