realsenseai/librealsense #899: can't get extrinsic information between motion module and color stream.

zhuzunjie17, 2017-12-11:
| Required Info  |   |
|---|---|
| Camera Model | ZR300 | 
| Firmware Version | 2.0.71.28  | 
| Operating System & Version | ubuntu 14.04  |
| Kernel Version (Linux Only)  |  4.4 |
| SDK Version | 1.12.1 |

I can't get extrinsic information between motion module and color stream.
while i use `get_motion_extrinsics_from(rs::stream::color)`, it return **rs.warn: Motion extrinsics validation from COLOR failed, because the data is invalid**. Does that means it don't offer extrinsics between motion module and color sensor?

dorodnic, 2017-12-11:
Hi @zhuzunjie17 
Please take a look at [#537](https://github.com/IntelRealSense/librealsense/issues/537).
This seem to be a known issue with the ZR300.

zhuzunjie17, 2017-12-12:
@dorodnic as you said in #537, get extrinsics Color to MM by multiplying the relavant two extrinsics will 
cause a degradation in precision, so is there another way i get a more precise extrinsics?

dorodnic, 2017-12-25:
Unfortunately, this seem to be the only work-around for this issue since the relevant tables are empty

closed 2017-12-31 by dorodnic