realsenseai/librealsense #901: d430 "No Frames Received!" in "Intel RealSense Viewer v2.8.3EXE"

tayzerker, 2017-12-11:
RealSense d430 shows "No Frames Received!" error continuously on windows when running the .exe straight off the get started page (https://software.intel.com/en-us/realsense/d400/get-started).  Device is rocognized (including firmware, etc.)  Tried all of my USB 3.0 ports, but no luck.  Any ideas?

| Required Info  |   |
|---|---|
| Camera Model | D430 | 
| Firmware Version | 05.08.06.00 | 
| Operating System & Version | Windows 10 |
| SDK Version |2.0 v2.8.3 |

dorodnic, 2017-12-12:
Hi @tayzerker 
This is not one of the common issues we encounter. Could you please share screenshots from Device Manager, Imaging Devices section (you can drag & drop images to GitHub comments) and camera info section of the Viewer? Using [this link](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q&A#q-how-do-i-enable-librealsense-logs) you can enable librealsense logs and share any them as well. Also, what build of Windows 10 are you using (you can find the build number under System Information)

tayzerker, 2017-12-12:
Thanks, @dorodnic .  This is all of the info I can think of.  Originally, I thought it might be because it was a USB 2.0 cable, so I got the best one I could find USB 3.1, but still no frames...

CURRENT SETUP:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

dorodnic, 2017-12-12:
The device was still identified as USB2, even when connected to a USB3 port. While you might be able to get some streaming via USB2 (with newer firmware) this is not the way we want you to experience the product. 
Do you have some way to check that your USB3 ports are functioning (perhaps a blue-marked USB stick. You should not see the "Device can perform faster" warning if the USB controller is in fact working properly). Alternatively, do you have another machine with USB3 to connect the camera to?

tayzerker, 2017-12-13:
I have actually finally gotten it to work!  For some reason the devices connect as USB 2.0 sometimes and as USB 3.0 other times - not sure exactly why though...    Thank you for your help!!

tayzerker, 2017-12-13:
Also, you mentioned newer firmware - I haven't been able to find instructions anywhere about how to update the firmware on the github pages or in the forum - do you happen to have a link or an idea of how to update?  Looks like I'm about 7 versions behind...

dorodnic, 2017-12-13:
Glad to hear it worked!

As a random guess I would try to explore USB settings in BIOS. On some boards the auto-detection is not working very well, and you can explicitly specify USB3 preference. Unfortunately, I don't really have a good reference for this, just from personal experience with some of the machines at the office. 

The firmware update protocol for Intel RealSense is somewhat proprietary while the SDK is open-source, so we can't bundle them together. However, with large number of engineering-samples out there we acknowledge the need for such tool. It is now being validated and is scheduled to be released soon for Windows and Linux. Once published we will link to it from GitHub documentation. 
If you can reach out to the Intel representatives who provided you the camera, they can most-likely send you pre-release software.

closed 2017-12-13 by dorodnic