realsenseai/librealsense #902: [D430] rs-depth: The connected device doesn't support depth streaming!

jfhamlin, 2017-12-11:
| Required Info  |   |
|---|---|
| Camera Model | D430 | 
| Firmware Version | 05.08.06.00 | 
| Operating System & Version | Ubuntu 14.04 |
| Kernel Version (Linux Only)  | 4.4.0-103-generic |
| SDK Version | 2.8.3 (current master@`0cc659d`) |

I've built the SDK myself, and when I run with the D430, I first see:

```
There are 1 connected RealSense devices.

Using device 0, an Intel RealSense USB2
```

Already, "USB2" seems suspicious. Then,

```
The connected device doesn't support depth streaming!
```

I updated rs-depth.c to include the error message, which is:

```
Failed to resolve request. No device found that satisfies all requirements
```

ev-mp, 2017-12-13:
Hi @jfhamlin,
Did you find why the device was not recognized as USB2 ? Was it due to power or cable issue ? 
Please update and provide logs, specifically:
__**>dmesg -wT**__   -  when connecting the camera to host
__**>lsusb**__  - after the device is connected

jfhamlin, 2017-12-13:
Thanks for the quick response. I believe the cable was at fault, but I'll try and get back to you.

jfhamlin, 2018-01-16:
Cabling was the issue. Thanks

closed 2018-01-16 by jfhamlin