realsenseai/librealsense #904: Raspberry Pi 3 not detecting D400 camera

armaghank, 2017-12-12:
| Required Info  |   |
|---|---|
| Camera Model | D400 | 
| Firmware Version |   | 
| Operating System & Version |  Ubuntu Mate 16.04 |
| Kernel Version (Linux Only)  |   |
| SDK Version | librealsense |

I have cloned and compiled the library successfully. However, all the tools or examples fail to detect the camera. The ports are working and so is the camera with Windows 10.

MichaelHanssonPH, 2017-12-12:
@armaghank, the D400 requires a USB 3.0 interface and the Raspberry Pi 3 only supports USB 2.0.

armaghank, 2017-12-12:
@MichaelHanssonPH I am following this document from the official repo https://github.com/IntelRealSense/librealsense/blob/master/doc/RaspberryPi3.md. It should work at a lower res.

dorodnic, 2017-12-12:
Hi @armaghank 
You are using a pre-production sample that is likely to have relatively old firmware. Can you please share the firmware version?

armaghank, 2017-12-12:
Hi @dorodnic yes it is a pre-production sample of 430.  The firmware version is 05.07.05.00

armaghank, 2017-12-13:
@dorodnic  I tried out the camera today with **Upboard** with a **USB 3.0 OTG** (running Ubuntu 16.04.3 LTS) and even that isn't detecting the camera after I compile the example codes. **Do you think that might be because of the firmware as well?** 
Btw the camera works fine with Linux on a normal laptop.

dorodnic, 2017-12-14:
No, it should work on the Upboard. I will check if there is any BIOS configuration required. We can help you troubleshoot the problem. Anything in the `dmesg` log when you connect the camera?

armaghank, 2017-12-19:
@dorodnic I got a D415 with a newer firmware **05.08.09.00** . In windows the realsense-viewer can detect it over a USB 2.0 connection. Raspberry Pi also detects the device but it throws this error: 
**rs_error was raised when calling rs2_create_device(info_list:0xa43dd0, index=0): Opcodes do not match! Sent 16 but received -1!** 
I will try it out with the UPBoard and let you know the result.

icarpis, 2017-12-31:
@armaghank, could you please [update](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#video4linux-backend-preparation) the udev rules on the Reaspberry PI and check if it solved the issue?
1. Ensure that no RealSense camera is connected
2. Copy the udev file to rules.d folder by typing `sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/`
3. Reload udev rules by typing `sudo udevadm control --reload-rules && udevadm trigger`

dorodnic, 2018-01-13:
@armaghank any update on this issue?

closed 2018-01-18 by dorodnic

BGA3, 2018-03-05:
Hi all

I also have this issue on my D415. I have updated the firmware to the newest version (5.9.2.0), and have also performed what @icarpis wrote in his last comment.

Do you guys have help that I can try?

Thanks in advance

FurqanHabibi, 2018-03-09:
Same problem here. Tried both D435 and D415 on firmware 5.8.15.0 and 5.9.2.0 and both are not detected over USB2 on Raspberry Pi and a linux desktop. They did get detected on the same linux desktop over USB3. Any help would be very appreciated!

Thank you.

FurqanHabibi, 2018-03-09:
Finally found a workaround for this. For anyone out there looking for a way to connect their realsense via USB2. The way to do it is:
1. First connect your USB type A end of the cable to your PC / raspberry pi
[image]
2. Then connect the type C end of the cable to the realsense
[image]
3. Check if realsense is detected or not using `lsusb`. It will show something like below if detected.
[image]
4. If it doesn't show up, repeat from step 2 (connecting type C end). For me it always shows up after the first or second try.

Obviously, this is not the proper way to connect realsense to your PC, so if someone from Intel can look into this, that will be greatly appreciated.

Cheers!

BGA3, 2018-03-13:
Hi @FurqanHabibi  

Thanks for your input. Unfurtunately, for me it does not resolve the issue. I always get the "opcode error", even if the device is listed using lsusb. 

Any help is greatly appreciated.

FurqanHabibi, 2018-03-15:
Hi @BGA3 

Yes, me too actually. I also got the "opcode error" even if it's listed by lsusb. Thankfully I found a way to fix this a few days ago. Turns out the USB port of raspberry pi doesn't provide enough power for realsense. So to fix that, I use an externally powered USB hub like this one:
[image]

Now the realsense is fully detected by raspberrypi and the SDK and examples can be run successfully. Keep in mind though that the realsense capabilities are severely limited by using it through USB2.

Hope it helps!

BGA3, 2018-03-16:
Sounds great, @FurqanHabibi! Thanks for the input.

I will try this out as soon as I get a hold of a unit