realsenseai/librealsense #906: What's difference between Intel RealSense SDK 2016 R2 with this project librealsense

fouvy, 2017-12-14:
I have noticed [here](https://software.intel.com/sites/landingpage/realsense/camera-sdk/v1.1/documentation/html/index.html?doc_devguide_introduction.html) that there is an old version document about Realsense SDK which has a lot of samples I can use. But this project librealsense 2.0 just have a few samples. My question is what's difference between these two projects? Where I can find document to compare these two SDK? I want reuse old samples code in librealsense 2.0 SDK.

| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version |   | 
| Operating System & Version |   |
| Kernel Version (Linux Only)  |   |
| SDK Version | |

dorodnic, 2017-12-14:
Hello @fouvy 

In a lack of better public documentation I can offer a list of key differences.

First of all, both libraries will work with the SR300 depth camera, but Intel RealSense SDK 2016 R2 is in process of being deprecated and we encourage customers to consider SDK 2.0.

That said, 

* Intel RealSense SDK **2016 R2** includes several computer-vision middlewares tailored for the depth camera. The algorithms are closed-source proprietary IP.
* Intel RealSense SDK **2.0** does not include any computer-vision middlewares. We will continue to provide samples and guides for integration with existing 3rd-party libraries and technologies ([like OpenCV](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv)), publish white-papers and end-to-end tutorials ([example here](https://software.intel.com/en-us/articles/build-an-autonomous-mobile-robot-with-the-intel-realsense-camera-ros-and-sawr)), and encourage community involvement. 
> As a result of extra functionality Intel RealSense SDK **2016 R2** has much larger footprint

* Intel RealSense SDK **2016 R2** was only validated on Windows 10, not including the later Microsoft updates (like the Creators Update). To the best of my knowledge it will not work on Windows 8, and certainly not on Linux or Mac OS. In addition, it requires relatively modern Intel CPU and will not run on ARM. 
* Intel RealSense SDK **2.0** is currently being tested on modern versions of Windows 10 and Ubuntu 16, but it will also work on earlier versions of Windows (Windows 8 and with some modifications also Windows 7), Mac OS, most Linux distributions and rooted Android phones. 

> Non-Intel specific optimizations are relatively low in our backlog, but such enhancements would be welcomed and we hope to offer more of them down the line

While the feature-set of the SDK had to be reduced, we are trying to increase the overall number of ways our best devices can be used.

closed 2017-12-19 by jlgarcia75

RealSense-Customer-Engineering, 2018-02-27:
[Realsense Customer Engineering Team Comment] 
Hi @fouvy
Do you still need help with this issue?
Regards