realsenseai/librealsense #909: rs2::sensor should have functinallity to get the profiles of currently opened streams

sogartar, 2017-12-14:
There is the function rs2::sensor::get_stream_profiles(), that returns the supported profiles that a sensor can provide.
There should be a function that returns the stream profiles of currently opened streams.

dorodnic, 2017-12-14:
That makes sense. 
There are several small enhancements to the API we are planning to add before D400 launch next month, we will add this to the list

RealSense-Customer-Engineering, 2018-05-15:
[Realsense Customer Engineering Team Comment]
@sogartar  By UVC protocol, the camera is the slave and the host is the master. The current camera stream profile is configured by host app. The host app should know what exactly is streaming. So there is no need a function to return the current profile.

closed 2018-05-24 by RealSense-Customer-Engineering