realsenseai/librealsense #910: http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/xenial/InRelease  throws not Authorized

alexrand, 2017-12-14:
trying to install according to https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

throws

<Error>
<Code>AccessDenied</Code>
<Message>Access Denied</Message>
<RequestId>423B9A9533062E28</RequestId>
<HostId>
Y6dSSdtt86UapGm0YtWWYR/DtOtiysmgu2jTJwOMobOcZdeKw3XYVQXVxax5eRTF3o3DKBzGmSA=
</HostId>
</Error>|

dorodnic, 2017-12-14:
Hi,
This seem to be a duplicate of #777, are you using Ubuntu 17? 
For the moment, we offer Debian packages for Ubuntu 16 (xenial) and for other distributions you can build the library from source.

alexrand, 2017-12-15:
yes , using ubuntu 17.10 - thnx, i did a build from sources.

RealSense-Customer-Engineering, 2018-02-28:
[Realsense Customer Engineering Team Comment] 
@**alexrand, would like to know if you still need help with this issue? **

dorodnic, 2018-03-19:
For the time being we can only offer Debian packages for Ubuntu 16.04 LTS. We will try to enable more distributions in the future.

closed 2018-03-19 by dorodnic