realsenseai/librealsense #912: How change the size of the color stream with the sdk 2.0

BernFlo, 2017-12-15:
| Required Info  |   |
|---|---|
| Camera Model | SR300  | 
| Firmware Version |   | 
| Operating System & Version |  Ubuntu 16.04 |
| Kernel Version (Linux Only)  | 4  |
| SDK Version |2.0 |

Hello,

I need a 480*640 color image for  my application but when I use data.get_color_frame() the size of the color image is 1080*1920. With the old sdk version I used pxcSenseManager->EnableStream(PXCCapture::STREAM_TYPE_COLOR, frameSize.width, frameSize.height, frameRate);
with Size frameSize = cv::Size(640, 480);
Do you know the equivalent for the sdk 2.0?

Thank you for your answer

dorodnic, 2017-12-15:
I believe this is what you need:
```cpp
pipeline pipe;
config cfg;
cfg.enable_stream(RS2_STREAM_COLOR, 640, 480);
// if you need depth - cfg.enable_stream(RS2_STREAM_DEPTH);
pipe.start(cfg);
```

BernFlo, 2017-12-15:
Yes Thank you very much. It's exactly what I need.

closed 2017-12-17 by dorodnic