realsenseai/librealsense #913: realsense-uvcvideo package installation fails on Ubuntu 16.04.3 with kernel 4.10

sogartar, 2017-12-15:
| Required Info  |   |
|---|---|
| Camera Model | R200 / F200 / SR300 / ZR300 / D400 | 
| Firmware Version |   | 
| Operating System & Version | Ubuntu 16.04.3 LTS |
| Kernel Version (Linux Only)  | 4.10.0-40-generic |
| SDK Version | |

I was following the [installation guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md).
On the step
```
sudo apt-get install realsense-uvcvideo
```
I got
```
The following NEW packages will be installed:
  realsense-uvcvideo
0 upgraded, 1 newly installed, 0 to remove and 54 not upgraded.
Need to get 79.0 kB of archives.
After this operation, 814 kB of additional disk space will be used.
Get:1 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial/main amd64 realsense-uvcvideo all 1.1.0-0ubuntu3 [79.0 kB]
Fetched 79.0 kB in 0s (175 kB/s)             
Selecting previously unselected package realsense-uvcvideo.
(Reading database ... 416035 files and directories currently installed.)
Preparing to unpack .../realsense-uvcvideo_1.1.0-0ubuntu3_all.deb ...
Unpacking realsense-uvcvideo (1.1.0-0ubuntu3) ...
Setting up realsense-uvcvideo (1.1.0-0ubuntu3) ...
Loading new realsense-uvcvideo-4.4.0 DKMS files...
First Installation: checking all kernels...
Building only for 4.10.0-40-generic
Building initial module for 4.10.0-40-generic
Error! Application of patch 15-realsense-camera-formats-4.10.patch failed.
Check /var/lib/dkms/realsense-uvcvideo/4.4.0/build/ for more information.
Loading new realsense-uvcvideo-4.10.0 DKMS files...
First Installation: checking all kernels...
Building only for 4.10.0-40-generic
Building initial module for 4.10.0-40-generic
Done.

uvcvideo:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.10.0-40-generic/updates/dkms/

depmod.....

DKMS: install completed.
```
Technically the package installed successfully. But I am don't think this is really true due to this part:
```
Error! Application of patch 15-realsense-camera-formats-4.10.patch failed.
Check /var/lib/dkms/realsense-uvcvideo/4.4.0/build/ for more information.
```
[This comment](https://github.com/IntelRealSense/librealsense/issues/444#issuecomment-321550925) mentions that Ubuntu 16.04.2 is supported. Is Ubuntu 16.04.3 supported?
I will try building from source.

ev-mp, 2017-12-15:
Hello @sogartar,
DKMS supports 16.04.3 with kernel 4.10.0-xx.
The internal build error is for DKMS trying to apply to kernel 4.4 (Ubuntu 14/16.04.1) in addition to 4.10.
Check that you've got the modified uvcvideo module for 4.10 according to the manual and update.

 I will add a task to review and update the package as well

sogartar, 2017-12-15:
I was successful when building from source. The D435 is running fine. :)

mjscosta, 2017-12-21:
Hi, I'm a bit confused with the librealsense2 support.
I'm having the same issue, but now when I install the package linux-headers-4.4.0-87-generic, I get an error:

`Setting up linux-headers-4.4.0-87-generic (4.4.0-87.110) ...
Examining /etc/kernel/header_postinst.d.
run-parts: executing /etc/kernel/header_postinst.d/dkms 4.4.0-87-generic /boot/vmlinuz-4.4.0-87-generic
Error! Application of patch 10-realsense-camera-formats-4.4.patch failed.
Check /var/lib/dkms/realsense-uvcvideo/4.10.0/build/ for more information.
`
1- Also, when compiling from source, the ubuntu-xenial git repo that gets checked out does not have the correct tag related to my version of the kernel, it has the latest (linux-image-4.4.0-104-generic), can't this create issues in the kernel?

2- Which kernel is supported? 

3- Do I really have to compile from source if I want to use the cameras ?

4- What if I want to distribute an app using realsense cameras, in linux ?

Thanks for your response in advance.

dorodnic, 2017-12-22:
The answer is somewhat different for D400 series and the SR300.
For the SR300, you need a Kernel version that supports the device. As a work-around you could build the library with `-DFORCE_LIBUVC=true` to handle everything in user-space, but generally speaking you want a proper kernel driver. 
The way to get this is either upgrade to a Kernel version that supports the device (4.12 and up), or deal with manual Kernel patching. We offer patches specialized for 4.4, 4.8 and 4.10 kernels, and scripts that we try to keep up to date. This is an uphill battle since every distribution is switching Kernel versions and adding custom patches all the time. 
In addition we offer `realsense-uvcvideo` DKMS package that we test regularly with out-of-the-box Ubuntu 16 and fix when necessary. Any changes you make to the machine (like installing additional Kernel version) are likely to require additional troubleshooting. We are happy to assist in such cases, to the extent that we can. 
The D400 series will be recognized out of the box by any modern Kernel (since 4.4.0.59) and only the advanced functionality ([UVC metadata](https://github.com/IntelRealSense/librealsense/blob/master/doc/frame_metadata.md#metadata-support-for-intel-realsense-devices)) will require a Kernel patch. Like with the SR300, full device support will eventually be a part of the Linux kernel, but these things take time. 

You are making valid points about redistribution and the complexity of building from source, but we are doing what we can to improve the situation. During 2017, we have upstreamed SR300 support, provided pre-compiled DKMS, significantly expanded [installation.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) and added [troubleshooting WIKI](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q&A).

mjscosta, 2017-12-25:
Hi @dorodnic , I'm new using the lib, I understand the points you mention. Thanks for hints and references!

chadrockey, 2018-01-08:
@dorodnic 
Could you provide a little more insight into these options you mention?  I have the D435 cameras and I'm trying to run on Ubuntu 14.04.

What functionality is supported since 4.4.0-59?  I'm on 4.4.0-103 and my camera doesn't connect unless I patch the kernel module.  It tries to connect for a long time, but eventually throws an rs2::error: "Failed to resolve request.  No device found that satisfies all requirements".  (rs2::error: "Could not find requested sensor type!" when I only request the color image stream).

Is there specific features or streams that can be used?  Can you request a depth stream without the patched module?  The rs-depth c examples reports to me that "The connected device doesn't support depth streaming!"

Also, when I build with -DFORCE_LIBUVC=true, I get printouts that say "unsupported descriptor subtype: 13", followed by the program sending a seg fault.  So that route doesn't seem to work as well, I'm just trying to understand the options available if you can't patch your kernel.

ev-mp, 2018-01-15:
Hello @chadrockey ,
In order to stream the depth data only you don't have to apply the patch, only the udev rules.
This will ,however, limit the device capabilities, and to take advantage of certain features the device provides you need the patches applied (see below).
We do not recommend using Librealsense with a live depth camera without patch/dkms applied.

As for the message you receive - please run "realsense-viewer" application and try to stream the depth sensor and report the results to better understand the nature of the error

With regards to the 4.4.0-59 - it provided for the Depth formats (Z16/Y8) to be streamed out of the box without applying patches.
This is sufficient to stream and receive the depth data, but in order to use advanced properties, such as metadata or motion tracking data (where appropriate) you need to apply the additional patches.

chadrockey, 2018-01-15:
Thanks @ev-mp I'll probably abandon running without a kernel patch given your recommendation.

When using the realsense-uvcvideo package on Ubuntu 14.04 I'm seeing the same issue as above.  Looks like the package installs 4.4 and 4.10 kernel patches and tries to build them both using dkms.  This succeeds in loading the module when you initially install the package, but when dkms is triggered on kernel updates, then it fails to load a patched module.

To get kernel updates working with DKMS, I removed the 4.10 files: sudo rm-rf /usr/src/realsense-uvcvideo-4.10.0 and sudo rm -rf /var/lib/dkms/realsense-uvcvideo/4.10.0

After removing these two directories, dkms triggers successfully when later kernel versions are installed.  Rebooting into the new kernel verifies that the patch and module load successfully.

Should there be separate debian packages for Ubuntu 14.04 and 16.04?  Is there a filter or installation rule in the dkms.conf that isn't working?  How do we disable the dkms rule that we do not need on our systems?

RealSense-Customer-Engineering, 2018-02-28:
[Realsense Customer Engineering Team Comment] 
Hello @chadrockey,

Do you still need help with this issue?

Regards,
Jesus G.
Intel Customer Support

chadrockey, 2018-02-28:
@RealSense-Customer-Engineering Can you have someone check the Debian for the dkms and verify the scripts?  When was the last time the Debian was released?  I haven't received an update so I believe this problem still exists.

mjscosta, 2018-02-28:
I have it working on Ubuntu 16.04, kernel 4.10 and 4.4, out of the box with following the installation instructions found here:
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

Make sure the kernel is one of 4.10 or 4.4, if you install latest ubuntu 16.04, you'll probably have to downgrade the kernel from 14.13 to 4.10 or 4.4...

mjscosta, 2018-02-28:
I cannot confirm.
I can only confirm that upgrading the librealsense2 works (e.g. 2.9 to 2.10), but not sure if that triggers the dkms.

I've downgraded yesterday, from 14.13 to 14.10, and had no problems, I mean that error pops up now and then (I started ignoring it), but it installs the modules and the camera works.

- Check with command `dkms status` if its installed.
- Can you try the camera despite of the error?

- my local output from `dkms status` (based on it, I would say it works):

```
realsense-uvcvideo, 4.4.0, 4.4.0-109-generic, x86_64: installed
realsense-uvcvideo, 4.4.0, 4.4.0-112-generic, x86_64: installed
realsense-uvcvideo, 4.4.0, 4.4.0-116-generic, x86_64: installed
```

RealsenseWilliam, 2018-05-03:
FYI ,  just  installed the Ubuntu 16.04.3 LTS, the kernel with it is 4.10.0-28 -generic 
used the DKMS script post on GitHub (Mar 1st version which still the latest today), 

use dkms status command to confirm the dkms installation 
>>dkms status
librealsense2-dkms, 1.2.0, 4.10.0-28-generic, X86_64 :installed

mofinfo show the uvcvideo is updated 
>>modinfo uvcvideo | grep : "version"
version :     1.1.2.realsense-1.2.0

run the realsense-viewer, D410 can streaming properly and able to configure the advanced controls 
not see issue on my setup

chadrockey, 2018-05-03:
Yes I can confirm this works now.  At some point the dkms package was reconfigured and works differently now.

closed 2018-05-10 by RealSense-Customer-Engineering