realsenseai/librealsense #914: Not detecting ZR300 on Intel Aero hardware USB OTG issue

ChrisHughes, 2017-12-15:
Camera: ZR300
Firmware: Unknown (as shipped out of the box)
Ubuntu 16.04 LTS
Kernel 4.4
Librealsense 1.12.1

(Reposting as my original issue on the matter was closed unexpectedly without resolution)

ZR300 connects and operates as expected on Ubuntu 16.04 on a laptop, however on the same OS, kernel on the Intel Aero Compute board, the camera is not detected (a small number of messages in dmesg, I'll post along with firmware when I'm back in the office on Thursday).

I am using a USB3.0 OTG cable to connect the camera, and have tried over both a powered USB hub as well as directly connected to the USB port. All other USB 3.0 devices tried function as expected.

Given the same OS, kernel and librealsense work on a laptop, my suspicions fall on the USB controller configuration in the InsydeH2O EFI bios. I'm unfamiliar with the configuration of this BIOS and have posted a screenshot in the hopes that someone might know what configuration options would help in establishing a working connection to the device.

[image]

Any help on the issue would be appreciated.

---REPLY 1---
The ZR-300 does not come with a 10-pin IPEX connector as the R-200 did. I am therefore trying to connect this via the USB 3.0 OTG port on the side of the Intel Atom. While I have the correct cables to connect this, it does not appear to be detecting.

I do not have the facility to create a custom IPEX to USB 3.0 OTG plug as I am on the road currently but need this up and running. Is there any BIOS settings I can modify so that the device is detected via this means of connecting it?

---UPDATE ON ORIGINAL POST---
In addition to USB3.0 issues with the RealSense, the Aero Compute board in question is having issues with any USB3.0 device such as flash drives connecting via USB3.0 OTG. If I plug the USB3.0 Flash Drive in via a downgraded 2.0 extension cable, it works just fine. This leads me to believe that the bios settings in the above screenshot are significant to identifying the problem and its solution.

When the RealSense device is plugged in to a computer it works fine on, I get the following four device IDs

```
Bus 002 Device 003: ID 8086:0ad0 Intel Corp. 
Bus 002 Device 004: ID 8086:0acb Intel Corp. 
Bus 002 Device 002: ID 0451:8025 Texas Instruments, Inc. 
Bus 001 Device 006: ID 0451:8027 Texas Instruments, Inc.
```
however on the Aero Compute board, I'm only getting the last of these four in an lsusb

```
Bus 001 Device 006: ID 0451:8027 Texas Instruments, Inc.
```

ChrisHughes, 2017-12-15:
I've found this forum post as well regarding the same issue I'm having https://communities.intel.com/thread/112263

jlgarcia75, 2017-12-15:
Hello @ChrisHughes,

Please post your issue in the Aero Community at https://communities.intel.com/community/tech/intel-aero. They will be able to answer your question there.

Regards,
Jesus G.
Intel Customer Support

ChrisHughes, 2017-12-17:
Thanks @jlgarcia75 I have made this post with the most current troubleshooting info on that site https://communities.intel.com/message/514685#514685
Please take a look if you can.

closed 2017-12-18 by jlgarcia75