realsenseai/librealsense #916: SR300 not working on mac

holstenholsten, 2017-12-17:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | 3.15.0.0  | 
| Operating System & Version | macOs 10.12.6  |
| SDK Version | 2.8.2 |

Hi!
I can run some of the examples for a short while before the system crashes. For instance, when running ./rs-depth, the error that is prompted is "unsupported descriptor subtype: 13". This descriptor subtype corresponds to UVC_VS_COLORFORMAT in libuvc.h. The is no handling of this subtype in device.c, hence the error. 

After 5 seconds I get the following error:
rs_error was raised when calling rs2_pipeline_wait_for_frames(pipe:0x7fdc4fc116c0):
    Frame didn't arrived within 5000

I have tried with two different cameras on two different macs, so it doesn't appear to be a hardware fault. Any idea what might be the problem and how to fix it?

Much appreciated 
Fredrik Holsten

[image]

tchapi, 2017-12-19:
Hello all,
I have the same "unsupported descriptor subtype: 13" log on a slightly equivalent setting :

| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | 3.21.0.0  | 
| Operating System & Version | macOs 10.12.6  |
| SDK Version | _2.8.3_ |

The frame does not arrive in time too : 

    rs2::error was thrown when calling rs2_pipeline_wait_for_frames(pipe:0x7fa0bb424450):
    Frame didn't arrived within 5000

I also have : 

    19/12 11:20:01,614 ERROR [0x7fffc7ec53c0] (types.h:123) fail to start streaming. 
    Last Error: Operation timed out

I'm using a homebrew bottled version of **librealsense**.
When I use the realsenseviewer, everything works perfectly, hence I think there might be some adjustements made in the viewer that needs to go in the docs for macOS... 

Looking through the issues I found that in https://github.com/IntelRealSense/librealsense/pull/752, @dorodnic mentions the `-DHWM_OVER_XU=false` for macOS (the brewed version of librealsense is not built with this flag) but I'm not sure his comment applies to the SR300 or the newest scanners.

tchapi, 2017-12-19:
As well, it's mentioned here in the [latest release](https://github.com/IntelRealSense/librealsense/releases/tag/v2.8.3) that "coming up" is "full OS X support", which implies there are still some missing pieces for macOS

dorodnic, 2017-12-19:
Hi @tchapi and @holstenholsten 
`unsupported descriptor subtype: 13` should not be written to the log, but it doesn't seem to be related to `Frame didn't arrived within 5000`. 
I agree that Mac OS validation is still not very good on our part, but I have tested `v2.8.3` tag with SR300 (FW 3.21.0.0) on 10.13.1 and couldn't reproduce the problem (both `realsense-viewer` and `capture` example worked). I wasn't able to install the bottled version because of firewall complications.
`DHWM_OVER_XU` should not affect SR300.

tchapi, 2017-12-19:
As a side note @dorodnic, do you recommend using the `FORCE_LIBUVC=true` flag when building on macOS ? I guess we should (if I understand the Wiki correctly), but the bottled version does not use the flag so I'm wondering ...

holstenholsten, 2017-12-19:
I tried v2.8.3 with macOS 10.13.2, and the same problems persists. When running rs-capture I get this:
[image]

When I start the realsense-viewer, I get this:
[image]

I can stream colour, at least for a while. When trying to stream depth, the camera disconnects. 

I built with Xcode, using the instructions from https://github.com/IntelRealSense/librealsense/blob/v2.8.3/doc/installation_osx.md. Did you do anything differently @dorodnic ?

dorodnic, 2017-12-20:
`FORCE_LIBUVC=true` will be enabled automatically (in lack of better backend)
I am also building with Xcode.
@holstenholsten, the problem you describe sounds more like a hardware issue (there are many similar reports on low-end Linux machines, but never so far on a Mac)
In particular, the camera will reset if it doesn't receive sufficient power (as result `No such file` error). This will not happen with RGB, since depth-generating ASIC is not doing much there, but can happen once you enable Depth.
What you could try is reducing the resolution / FPS or connecting through powered USB3 Hub. In addition, I would also double-check the cables (both USB3 OTG cable and the USB-C dongle if one is used).

xtrawurst, 2018-01-05:
I have the same issue on my Mac (10.12.6), but with a D415.

Required Info |
------------ | -------------
Camera Model | D415
Firmware Version | 05.08.09
Operating System & Version | macOS 10.12.6
SDK Version | 2.9.0

 The "Realsense Viewer" window opens but it gets stuck at the start screen with the blinking "Realsense device detected" and constantly printing `unsupported descriptor subtype: 13` to the console.

tchapi, 2018-01-25:
Any update on the `unsupported descriptor subtype: 13` log that is written to stdout ?

ProfFan, 2018-02-20:
My 5 cents on this too. Struggled for nearly two months with no outcome. My logs are attached below.

```
tools/realsense-viewer/realsense-viewer                ⬡ 6.8.0 [±master ✓]
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
 20/02 13:26:37,955 ERROR [0x700005f37000] (types.h:178) Could not open device. Last Error: Undefined error: 0
 20/02 13:26:37,955 WARNING [0x700005f37000] (sensor.cpp:683) Exception was thrown when inspecting properties of a sensor
 20/02 13:26:37,955 WARNING [0x700005f37000] (sensor.cpp:683) Exception was thrown when inspecting properties of a sensor
 20/02 13:26:37,955 WARNING [0x700005f37000] (sensor.cpp:683) Exception was thrown when inspecting properties of a sensor
 20/02 13:26:37,957 WARNING [0x700005f37000] (sensor.cpp:683) Exception was thrown when inspecting properties of a sensor
 20/02 13:26:37,957 WARNING [0x700005f37000] (sensor.cpp:683) Exception was thrown when inspecting properties of a sensor
 20/02 13:26:37,958 WARNING [0x700005f37000] (sensor.cpp:683) Exception was thrown when inspecting properties of a sensor
 20/02 13:26:37,958 WARNING [0x700005f37000] (sensor.cpp:683) Exception was thrown when inspecting properties of a sensor
 20/02 13:26:37,958 WARNING [0x700005f37000] (sensor.cpp:683) Exception was thrown when inspecting properties of a sensor
 20/02 13:26:37,959 WARNING [0x700005f37000] (sensor.cpp:683) Exception was thrown when inspecting properties of a sensor
 20/02 13:26:37,959 WARNING [0x700005f37000] (sensor.cpp:683) Exception was thrown when inspecting properties of a sensor
 20/02 13:26:37,959 WARNING [0x700005f37000] (sensor.cpp:683) Exception was thrown when inspecting properties of a sensor
[1]    30585 segmentation fault  tools/realsense-viewer/realsense-viewer
```

omarev-git, 2018-03-14:
I have an RS300 that work ok on Win10, but over macOS Sierra/MacPro 2013 with lisrealsense 2.10.2. I try in two ways and both of them fail.
First plugged directly to MacPro, an RGB buffer can be acquired but not depth buffer. In depth request the SR300 simply reset himself.
Second using a hub usb 3.0 with electric power, here realsense-viewer example work I can retrieve rgb and depth buffer BUT only one of the buffers can be streamed each time, I mean, if depth buffer is readed cannot be readed rgb buffer.  This the output:

2018-03-14 18:08:39.375027-0400 realsense-viewer[1166:221844] [General] ERROR: Setting <GLFWContentView: 0x101c655d0> as the first responder for window <GLFWWindow: 0x101f20ef0>, but it is in a different window ((null))! This would eventually crash when the view is freed. The first responder will be set to nil.
(
	0   AppKit                              0x00007fff9c6b99cf -[NSWindow _validateFirstResponder:] + 557
	1   AppKit                              0x00007fff9be3474c -[NSWindow _setFirstResponder:] + 31
	2   AppKit                              0x00007fff9bedc9ce -[NSWindow _realMakeFirstResponder:] + 406
	3   AppKit                              0x00007fff9bedc7e4 -[NSWindow makeFirstResponder:] + 123
	4   libglfw.3.dylib                     0x0000000101a31895 _glfwPlatformCreateWindow + 631
	5   libglfw.3.dylib                     0x0000000101a2d430 glfwCreateWindow + 487
	6   realsense-viewer                    0x00000001001a3fe0 _ZN3rs29ux_window11open_windowEv + 1808
	7   realsense-viewer                    0x00000001001a4bac _ZN3rs29ux_windowC2EPKc + 1916
	8   realsense-viewer                    0x00000001001a54dd _ZN3rs29ux_windowC1EPKc + 29
	9   realsense-viewer                    0x000000010020fc55 main + 85
	10  libdyld.dylib                       0x00007fffb3ef5235 start + 1
	11  ???                                 0x0000000000000001 0x0 + 1
)
unsupported descriptor subtype: 13
(.... same previous line...)
unsupported descriptor subtype: 13
 14/03 18:08:39,601 WARNING [0x70000591f000] (sensor.cpp:313) Unregistered Media formats : [ INVZ INZC PAIR YUY2 ]; Supported: [ ]
unsupported descriptor subtype: 13
(... same line)
unsupported descriptor subtype: 13
 14/03 18:08:49,576 WARNING [0x101ad73c0] (sensor.cpp:313) Unregistered Media formats : [ INVZ INZC PAIR YUY2 ]; Supported: [ ]
unsupported descriptor subtype: 13
(.... same line)
unsupported descriptor subtype: 13
 14/03 18:08:49,658 WARNING [0x101ad73c0] (archive.cpp:316) The user was holding on to 1 frames after stream 0x0x102239220 stopped
 14/03 18:08:49,658 WARNING [0x101ad73c0] (archive.cpp:316) The user was holding on to 1 frames after stream 0x0x102239220 stopped
 14/03 18:08:49,658 WARNING [0x101ad73c0] (archive.cpp:326) All frames from stream 0x0x102239220 are now released by the user

RealSense-Customer-Engineering, 2018-04-16:
[Realsense Customer Engineering Team Comment] 
hi @omarev-git,

I just try this on MacOS High Sierra (10.13.2), MacBook Air 2017 with Librealsense 2.10.3, plug the SR300 directly to Mac, which is working good. so please let me know if you still having same issue

RealSense-Customer-Engineering, 2018-04-16:
[Realsense Customer Engineering Team Comment] 
hi @holstenholsten,
for Frame didn't arrived within 5000, my previous experience was cause by USB cable, so please make sure you are using USB certified cable, and please let me know if you still have this issue with latest release v2.10.3, thanks.

RealSense-Customer-Engineering, 2018-05-02:
[Realsense Customer Engineering Team Comment]
hi @holstenholsten,

is the latest release works for you? do you need more support on this topic?
thanks.

RealSense-Customer-Engineering, 2018-05-11:
[Realsense Customer Engineering Team Comment]
let me close this ticket here, please feel free to open new ticket if any question.

closed 2018-05-11 by RealSense-Customer-Engineering

omarev-git, 2018-10-30:
I tray with the last release 2.16.1 and nothing, I think this is the trouble the 3d sensor is linked to USB2.0 HUB, why?
How I wrote in previous messages my mac  is MacBook Pro 2015 and both USB ports are 3.0. By the way my mac is attached to power source and the USB cable for the sensor is the cable that come with the sensor. May be some one have test sanity application?
[image]