realsenseai/librealsense #921: cannot get intrinsics for infrared stream

svensht2, 2017-12-19:
I'm using using RS415 with FW 5.8.14.0. librealsense 2.8.3, Windows 10 1709.
I'm enabling INFRARED stream and when I try to get instrinsics for this stream I get the message:
19/12 00:54:18,153 WARNING [12036] (types.cpp:57) No intrinsics are available for this stream profile!

I expect the INFRARED stream to have same instrinsics as the depth (which I can get successfully).

svensht2, 2017-12-19:
seems like its not throwing this error if I set the INFRARED stream to RGB8, but it does with Y16

ev-mp, 2017-12-19:
Hi @svensht2,
For D415 depth camera Infrared Y16 is a calibration (unrectified) format, and does not provide intrinsic by design.
Additionally, not all resolution/fps combination available for Y16 are supported by Z16 depth stream (e.g. FullHD/25fps).

svensht2, 2017-12-19:
what I really want is Y8, which 415 doesn't like at all, is that expected?

ev-mp, 2017-12-20:
Y8 is supported - you may run RealSense-Viewer and select this format for preview:
[image]
Note that in order to get both left and right Infrared streams you need to switch to "Advanced Mode" - see the button "Enable Advanced Mode" in the above image.

svensht2, 2017-12-22:
I will try using Y8 again

svensht2, 2017-12-24:
still having the Y8 issue. I'm experiencing the same issue as #937

closed 2017-12-25 by svensht2

freemanlo, 2017-12-28:
Tried to run below app (in librealsense 2.8.3) using D415 + FW 5.8.9 and the intrinsics is there. But, the coeffs are all '0'.

$ /rs-enumerate-devices -c

----------------------------------------------------
Provided Intrinsic:
Intrinsic of "Depth"	  424x240	  Z16
Width:        	424
Height:       	240
PPX:          	212.520492553711
PPY:          	116.628021240234
Fx:           	318.328033447266
Fy:           	318.328033447266
Distortion:   	Brown Conrady
Coeffs:       	0  	0  	0  	0  	0  

Intrinsic of "Depth"	  480x270	  Z16
Width:        	480
Height:       	270
PPX:          	240.585556030273
PPY:          	131.206527709961
Fx:           	358.119049072266
Fy:           	358.119049072266
Distortion:   	Brown Conrady
Coeffs:       	0  	0  	0  	0  	0  

Intrinsic of "Infrared 1"	  424x240	  Y8
Width:        	424
Height:       	240
PPX:          	212.520492553711
PPY:          	116.628021240234
Fx:           	318.328033447266
Fy:           	318.328033447266
Distortion:   	Brown Conrady
Coeffs:       	0  	0  	0  	0  	0  

Intrinsic of "Infrared 1"	  480x270	  Y8
Width:        	480
Height:       	270
PPX:          	240.585556030273
PPY:          	131.206527709961
Fx:           	358.119049072266
Fy:           	358.119049072266
Distortion:   	Brown Conrady
Coeffs:       	0  	0  	0  	0  	0