realsenseai/librealsense #925: rs2::align Issue

manhha1402, 2017-12-19:
| Required Info  |   |
|---|---|
| Camera Model |  SR300 | 
| Firmware Version | 3.15.0.0  | 
| Operating System & Version |  14.04 |
| Kernel Version (Linux Only)  |   |
| SDK Version | |
I updated librealsense2 today, recompiled the library and tested the example "rs-align". I found some lines in depth stream. I convert depth to point cloud and see as images. I think it has problem with the function "rs2::align". 
[image]
[image]

dorodnic, 2017-12-20:
I think this is an artifact of how the artificial image is generated. We probably need some sort of filtering, since right now it looks like "nearest" interpolation

jjoachim, 2017-12-28:
Just curious, how are you doing rs-align through the pcl_viewer?

closed 2018-01-29 by manhha1402