realsenseai/librealsense #927: Frame didn't arrived within 5000 with RS-D435 on Ubuntu14.04

ytzhao, 2017-12-20:
| Required Info  |   |
|---|---|
| Camera Model | D435
| Firmware Version |   05.08.14.00
| Operating System & Version |   Ubuntu14.04
| Kernel Version (Linux Only)  |   4.4.0-101-generic
| SDK Version | rs2

Hi everyone, I developed a project with RealSense D435 under Qt5.9. I read the color frame and depth frame data by rs2 api default function.  
 
The core code segment is as followed:
```cpp
while(1) 
{
    rs2::frameset frameset = pipeline.wait_for_frames();
    rs2::frame depth_frame = frameset.get_depth_frame();

    matDepth = cv::Mat(cv::Size(640,480), CV_16UC1, (void*)depth_frame.get_data());
    
    cv::imshow("depth img", matDepth);
    cv::waitKey(10);
}
```
The program could run correctly for several minutes, but **each time** it will terminate unexpectly with error message : 
**terminate called after throwing an instance of 'rs2::error'  
what(): Frame didn't arrived within 5000**

icarpis, 2017-12-21:
Hello @ytzhao, do you see the same behavior with RealSense samples/tools such as realsense-viewer or rs-depth-quality?
If so, could you please provide RealSense log by adding "log_to_file(RS2_LOG_SEVERITY_DEBUG);" to your app?

There is also a possibility that your camera hangs up while streaming. In order to ensure it doesn't happen please try to monitoring the USB connect/disconnect events by doing the following:
1) Open Terminal.
2) Type `udevadm monitor`
3) Try to reproduce the issue

If the camera hangs up you will see output from udevadm.

ytzhao, 2017-12-26:
Hello @icarpis , firstly, thanks for your reply. 

1. I tried with RealSense samples/tools/realsense-viewer, there are about 2 out of 10 times output:
```shell
- librealsense version: 2.8.2
- Intel RealSense 435 was selected as a default device
- Incomplete frame detected!
Size 328712 out of 614655 bytes (53%)
```
2. I added "log_to_file(RS2_LOG_SEVERITY_DEBUG);" to my app, the app ran for about 15 minutes and then crashed. The log file shows that: 
```text
 26/12 15:35:01,000 DEBUG [139909046900480] (archive.cpp:157) CallbackFinished,Color,0,DispatchedAt,1514273701000.158447
 26/12 15:35:01,000 DEBUG [139909046900480] (archive.cpp:164) Frame Callback [Color#0] overdue. (Duration: 1514273701000.158447ms, FPS: 30, Max Duration: 32ms)
 26/12 15:35:01,000 DEBUG [139909046900480] (sync.cpp:26) SYNCED: Color 21169, 884825.766000
Infrared 21343, 884834.943000
Infrared 21343, 884834.943000
Depth 21343, 884834.943000

 26/12 15:35:01,000 DEBUG [139909046900480] (archive.cpp:361) CallbackStarted,Color,21169,DispatchedAt,1514273701000.286865
 26/12 15:35:01,000 DEBUG [139909046900480] (archive.cpp:157) CallbackFinished,Infrared,21341,DispatchedAt,1514273701000.330566
 26/12 15:35:01,000 DEBUG [139909046900480] (archive.cpp:164) Frame Callback [Infrared#21341] overdue. (Duration: 68.186035ms, FPS: 30, Max Duration: 32ms)
 26/12 15:35:01,000 DEBUG [139909055293184] (archive.cpp:157) CallbackFinished,Infrared,21341,DispatchedAt,1514273700999.356201
 26/12 15:35:01,000 DEBUG [139909055293184] (archive.cpp:164) Frame Callback [Infrared#21341] overdue. (Duration: 66.955811ms, FPS: 30, Max Duration: 32ms)
 26/12 15:35:01,000 DEBUG [139909055293184] (archive.cpp:157) CallbackFinished,Color,21168,DispatchedAt,1514273701000.613525
 26/12 15:35:01,000 DEBUG [139909055293184] (archive.cpp:157) CallbackFinished,Color,0,DispatchedAt,1514273701000.657227
 26/12 15:35:01,001 DEBUG [139909046900480] (archive.cpp:157) CallbackFinished,Depth,21341,DispatchedAt,1514273701000.951904
```

3. I watched the device monitor, when my application runs, the camera does not hang up.

icarpis, 2017-12-28:
Thank you for the detailed outputs.
The prints of `Incomplete frame detected` means that either the machine or the USB controller are weak.
Regarding the realsense-viewer crash, please provide core dump file by the following steps:
1. Open a Terminal
2. Type `ulimit -c unlimited`
3. Reproduce the crash of `realsense-viewer`
4. A core file should be appears at the working directory
5. Revert back the core dump settings to default by typing `ulimit -c 0`

mjscosta, 2018-03-02:
Hi, I had this issue now and then.
@icarpis What do you mean with:
> The prints of Incomplete frame detected means that either the machine or the USB controller are weak.

Do you mean, it has power issues (not enough power for the device)? do you mean the controller has not enough bandwidth to transfer the full frames ?

dorodnic, 2018-03-25:
Sorry for delay, the partial frames issue usually indicate that the `uvcvideo` driver does not get enough `USB` buffers from the USB subsystem. This can be either because the system is overloaded (possibly thermal-throttled), CPU utilisation is near 100%, or the USB controller is not handling the requests in time.

RealSense-Customer-Engineering, 2018-05-07:
[Realsense Customer Engineering Team Comment]
hi @mjscosta , @ytzhao

do you still need support on this topic?
thanks.

RealSense-Customer-Engineering, 2018-05-21:
[Realsense Customer Engineering Team Comment]
let me close this ticket here, please feel free to open new ticket if any question.

closed 2018-05-21 by RealSense-Customer-Engineering

mjscosta, 2018-05-21:
Fell free to close it on my side. I've got this under a VM, either using vmware or virtualbox.
So dorodnic's reply, just explains it. Thanks for the info.

Best.