realsenseai/librealsense #929: ZR300 stream not usable

ABesginow, 2017-12-20:
| Required Info  |   |
|---|---|
| Camera Model | ZR300  | 
| Operating System & Version | Ubuntu 16.04  |
| Kernel Version (Linux Only)  | 4.10.0-42-generic  |

Following issue:

My ZR300 is suddenly not recognized by the realsense-viewer and python.

Running a script I get the following error message (http://opencv-python-tutroals.readthedocs.io/en/latest/py_tutorials/py_gui/py_video_display/py_video_display.html):
```
libv4l2: error set_fmt gave us a different result then try_fmt!
HIGHGUI ERROR: libv4l unable convert to requested pixfmt
```
_lsusb_ recognizes the device: (should be the three first entries)
```
Bus 002 Device 010: ID 8086:0ad0 Intel Corp. 
Bus 002 Device 011: ID 8086:0acb Intel Corp. 
Bus 002 Device 009: ID 0451:8025 Texas Instruments, Inc. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 8087:0a2b Intel Corp. 
Bus 001 Device 005: ID 046a:b090 Cherry GmbH 
Bus 001 Device 004: ID 046d:c077 Logitech, Inc. M105 Optical Mouse
Bus 001 Device 003: ID 058f:6387 Alcor Micro Corp. Flash Drive
Bus 001 Device 009: ID 0451:8027 Texas Instruments, Inc. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
> lsusb without camera
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 8087:0a2b Intel Corp. 
Bus 001 Device 005: ID 046a:b090 Cherry GmbH 
Bus 001 Device 004: ID 046d:c077 Logitech, Inc. M105 Optical Mouse
Bus 001 Device 003: ID 058f:6387 Alcor Micro Corp. Flash Drive
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

and dmesg shows that the camera exists and is detected right (I guess):

```
[ 1659.080234] usb 1-1: new high-speed USB device number 10 using xhci_hcd
[ 1659.220587] usb 1-1: New USB device found, idVendor=0451, idProduct=8027
[ 1659.220593] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=1
[ 1659.220597] usb 1-1: SerialNumber: 900F0849A8EF
[ 1659.221704] hub 1-1:1.0: USB hub found
[ 1659.221767] hub 1-1:1.0: 2 ports detected
[ 1659.340154] usb 2-1: new SuperSpeed USB device number 13 using xhci_hcd
[ 1659.360584] usb 2-1: New USB device found, idVendor=0451, idProduct=8025
[ 1659.360590] usb 2-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[ 1659.361832] hub 2-1:1.0: USB hub found
[ 1659.363056] hub 2-1:1.0: 2 ports detected
[ 1659.648364] usb 2-1.2: new SuperSpeed USB device number 14 using xhci_hcd
[ 1659.669261] usb 2-1.2: LPM exit latency is zeroed, disabling LPM.
[ 1659.670004] usb 2-1.2: New USB device found, idVendor=8086, idProduct=0ad0
[ 1659.670011] usb 2-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1659.670015] usb 2-1.2: Product: CX3-UVC
[ 1659.670019] usb 2-1.2: Manufacturer: Cypress
[ 1659.671291] uvcvideo: Found UVC 1.10 device CX3-UVC (8086:0ad0)
[ 1659.674350] uvcvideo 2-1.2:1.0: Entity type for entity Extension 3 was not initialized!
[ 1659.674361] uvcvideo 2-1.2:1.0: Entity type for entity Processing 2 was not initialized!
[ 1659.674368] uvcvideo 2-1.2:1.0: Entity type for entity Camera 1 was not initialized!
[ 1660.000494] usb 2-1.1: new SuperSpeed USB device number 15 using xhci_hcd
[ 1660.023589] usb 2-1.1: New USB device found, idVendor=8086, idProduct=0acb
[ 1660.023595] usb 2-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1660.023599] usb 2-1.1: Product: Intel RealSense 3D Camera ZR300
[ 1660.023602] usb 2-1.1: Manufacturer: Intel Corp
[ 1660.023605] usb 2-1.1: SerialNumber: SN_3511803250
[ 1660.026243] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera ZR300 (8086:0acb)
[ 1660.029632] uvcvideo 2-1.1:1.0: Entity type for entity Extension 2 was not initialized!
[ 1660.029642] uvcvideo 2-1.1:1.0: Entity type for entity Camera 1 was not initialized!
[ 1660.030728] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera ZR300 (8086:0acb)
[ 1660.033880] uvcvideo: Unable to create debugfs 2-15 directory.
[ 1660.034182] uvcvideo 2-1.1:1.2: Entity type for entity Camera 4 was not initialized!
[ 1660.035193] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera ZR300 (8086:0acb)
[ 1660.038356] uvcvideo: Unable to create debugfs 2-15 directory.
[ 1660.038638] uvcvideo 2-1.1:1.4: Entity type for entity Processing 7 was not initialized!
[ 1660.038646] uvcvideo 2-1.1:1.4: Entity type for entity Camera 6 was not initialized!
[ 1660.039015] input: Intel RealSense 3D Camera ZR300 as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1.1/2-1.1:1.4/input/input16
```

I reviewed all the solved and open issues and applied a number of patches and it is still not working.

Any help regarding this is appreciated.

dorodnic, 2017-12-24:
Hi @ABesginow 
At the moment, ZR300 is only supported by it's dedicated software stack:
* [Linux SDK](https://software.intel.com/sites/products/realsense/intro/getting_started.html)
* [librealsense 1.12.1](https://github.com/IntelRealSense/librealsense/tree/legacy)
Are you able to work with it using `cpp-config-ui` utility?

ABesginow, 2018-01-02:
That was exactly the issue.
It works using the `cpp-config-ui`.

Thanks a lot!

closed 2018-01-02 by ABesginow