realsenseai/librealsense #931: How to handle camera disconnection while streaming?

Jiloc, 2017-12-20:
| Required Info  |   |
|---|---|
| Camera Model |  SR300 / D400 | 
| Firmware Version |   | 
| Operating System & Version |  Windows 10  |
| Kernel Version (Linux Only)  |   |
| SDK Version | 2.8.3 |

I am writing my own interface on top of librealsense to handle the camera (only one camera active at any given time).

Everything seem to work like a charm, i receive the events when a camera connects or disconnects, i can start and stop a camera to send the depth frames.

The problems arise when I disconnect the camera while streaming. 
After that if I try to reconnect the same camera, the connection event is not fired but, if I connect and disconnect another camera they are fired as always.

To handle the connection/disconnection event I use:
    
     rs2::contenxt::set_devices_changed_callback

To enable the device I want to use I rely on:

    rs2::config config;
    config.enable_stream(RS2_STREAM_DEPTH);
    config.enable_device("someserialnumber");

To start streaming I use:

     rs2::pipeline pipe;
     pipe.start(config);

To generate the frames i use:

    rs2::frameset data = pipe->wait_for_frames(); 
    rs2::frame depth = data.get_depth_frame(); 

When I disconnect the camera:
1) I receive the disconnect signal
2) the programs is blocked for a few seconds waiting for new frames
3) the programs throws a rs2::camera_disconnected_error

What should I do when I receive the disconnection signal? 
Should I have to turn off the pipeline, the device, the sensor or whatever ?

Jiloc, 2017-12-21:
Hi @zivsha ,

I am using a a secondary thread to get the frames, and differently from the rs-multicam example I am not pulling but waiting for the next frame. 

I tried also pulling, but still doesn't work. 

Anyway I investigated further and it seems like the event is fired and the rs2::event_information::get_new_devices() returns a device_list with 1 element. (I can check it with .size()).

The problem is that what I try to access that member, it throws a rs2::backend_error

> "_source->QueryInterface(__uuidof(IKsTopologyInfo), reinterpret_cast<void **>(&ks_topology_info)) returned: HResult 0x80004002: \"Interfaccia non supportata.\""

I tried to access to it using the iterator:

    for (auto&& dev : devices)
    {
        addDevice(dev);
    }

But also directly:

    rs2::device dev = devices[0];

Jiloc, 2017-12-21:
If you are on windows and have qt installed you can execute the project just changing the Product ID of your Realsense [Here](https://github.com/Jiloc/tighty/blob/master/example/main.cpp)

[Here](https://github.com/Jiloc/tighty/blob/master/src/camera/realsense/rscameramanager.cpp#L29) you can see where the exception is captured

RealSense-Customer-Engineering, 2018-05-11:
[Realsense Customer Engineering Team Comment]
@Jiloc
I revised the code of rs-capture.cpp to suit your request, please let me know if it works for you.

Jiloc, 2018-05-17:
@carypai  thank you for looking into it.

I can confirm that your implementation works.

It just seems that the camera connection/disconnection handling it's not thread safe. It must be handled by the same thread that uses the pipeline to get frames.

RealSense-Customer-Engineering, 2018-05-18:
[Realsense Customer Engineering Team Comment]
@Jiloc
Can you describe more why do you think it seems not thread safe?

Pipeline is within the same context of even handling and when you get the device is disconnected, you can set the disconnect to true and not going into getting back frame. If it already goes into wait for frame, it's still okay, exception will throw out after time out.

closed 2018-05-24 by RealSense-Customer-Engineering