realsenseai/librealsense #933: SR300 and Jetson Tx2

mbayog02, 2017-12-21:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version |   | 
| Operating System & Version |  Jetpack 3.1 |
| Kernel Version (Linux Only)  |  4.4.38 |
| SDK Version | |

I've followed the installation guide for Librealsense in Lunux but when I try .examples//cpp-config-ui I get the following error code:
`nvidia@tegra-ubuntu:~/librealsense/build/examples$ ./cpp-config-ui 
Capturing DEPTH at 640 x 480, fov = 67.7 x 53.4, distortion = INVERSE_BROWN_CONRADY
Capturing COLOR at 640 x 480, fov = 54.1 x 41.9, distortion = NONE
Capturing INFRARED at 640 x 480, fov = 67.7 x 53.4, distortion = INVERSE_BROWN_CONRADY
terminate called after throwing an instance of 'std::runtime_error'
  what():  VIDIOC_DQBUF error 19, No such device
Aborted (core dumped)
`

I can see SR300 from lsusb but when I try dmesg I get that:

`[ 2335.642827] usb 2-1.3.3: new SuperSpeed USB device number 13 using xhci-tegra
[ 2335.670083] usb 2-1.3.3: New USB device found, idVendor=05e3, idProduct=0616
[ 2335.677218] usb 2-1.3.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 2335.684959] usb 2-1.3.3: Product: USB3.1 Hub
[ 2335.689481] usb 2-1.3.3: Manufacturer: GenesysLogic
[ 2335.700103] hub 2-1.3.3:1.0: USB hub found
[ 2335.704726] hub 2-1.3.3:1.0: 1 port detected
[ 2335.750068] usb 1-2.3.3: new high-speed USB device number 12 using xhci-tegra
[ 2335.861366] usb 1-2.3.3: New USB device found, idVendor=05e3, idProduct=0610
[ 2335.868501] usb 1-2.3.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 2335.876151] usb 1-2.3.3: Product: USB2.0 Hub
[ 2335.880613] usb 1-2.3.3: Manufacturer: GenesysLogic
[ 2335.888057] hub 1-2.3.3:1.0: USB hub found
[ 2335.892578] hub 1-2.3.3:1.0: 2 ports detected
[ 2336.730127] usb 1-2.3.3.2: new full-speed USB device number 13 using xhci-tegra
[ 2336.987364] usb 1-2.3.3.2: New USB device found, idVendor=041e, idProduct=409b
[ 2336.994693] usb 1-2.3.3.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2337.002653] usb 1-2.3.3.2: Product: BlasterX Senz3D VF0810
[ 2337.008356] usb 1-2.3.3.2: Manufacturer: Creative Technology Ltd
[ 2337.014528] usb 1-2.3.3.2: SerialNumber: 170523141351
[ 2337.177251] input: Creative Technology Ltd BlasterX Senz3D VF0810 as /devices/3530000.xhci/usb1/1-2/1-2.3/1-2.3.3/1-2.3.3.2/1-2.3.3.2:1.2/0003:041E:409B.0007/input/input22
[ 2337.247053] hid-generic 0003:041E:409B.0007: input,hidraw3: USB HID v1.11 Device [Creative Technology Ltd BlasterX Senz3D VF0810] on usb-3530000.xhci-2.3.3.2/input2
[ 2337.498602] xhci-tegra 3530000.xhci: tegra_xhci_mbox_work mailbox command 5
[ 2337.505570] xhci-tegra 3530000.xhci: tegra_xhci_mbox_work ignore firmware MBOX_CMD_DEC_SSPI_CLOCK request
[ 2337.575587] xhci-tegra 3530000.xhci: tegra_xhci_mbox_work mailbox command 4
[ 2337.582805] xhci-tegra 3530000.xhci: tegra_xhci_mbox_work ignore firmware MBOX_CMD_INC_SSPI_CLOCK request
[ 2337.798793] usb 2-1.3.3.1: new SuperSpeed USB device number 14 using xhci-tegra
[ 2337.824553] usb 2-1.3.3.1: New USB device found, idVendor=8086, idProduct=0aa5
[ 2337.832171] usb 2-1.3.3.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2337.840366] usb 2-1.3.3.1: Product: Intel(R) RealSense(TM) Camera SR300
[ 2337.847362] usb 2-1.3.3.1: Manufacturer: Intel(R) RealSense(TM) Camera SR300
[ 2337.854750] usb 2-1.3.3.1: SerialNumber: 627201003373
[ 2337.862302] xhci-tegra 3530000.xhci: tegra_xhci_mbox_work mailbox command 6
[ 2337.873697] uvcvideo: Found UVC 1.10 device Intel(R) RealSense(TM) Camera SR300 (8086:0aa5)
[ 2337.888362] input: Intel(R) RealSense(TM) Camera S as /devices/3530000.xhci/usb2/2-1/2-1.3/2-1.3.3/2-1.3.3.1/2-1.3.3.1:1.0/input/input23
[ 2337.902375] uvcvideo: Unknown video format 435a4e49-b602-480f-978c-e4e88ae89b89
[ 2337.909849] uvcvideo: Unknown video format 52494150-8536-4841-b6bf-8fc6ffb083a8
[ 2337.917777] uvcvideo: Found UVC 1.10 device Intel(R) RealSense(TM) Camera SR300 (8086:0aa5)
[ 2337.932392] uvcvideo: Unable to create debugfs 2-14 directory.
[ 2337.938893] input: Intel(R) RealSense(TM) Camera S as /devices/3530000.xhci/usb2/2-1/2-1.3/2-1.3.3/2-1.3.3.1/2-1.3.3.1:1.2/input/input24
[ 2342.578479] xhci-tegra 3530000.xhci: tegra_xhci_mbox_work mailbox command 5
[ 2342.585792] xhci-tegra 3530000.xhci: tegra_xhci_mbox_work ignore firmware MBOX_CMD_DEC_SSPI_CLOCK request
`

I've tried plugging in the camera to a external powered USB3 HUB and directly to the Jetson USB port but nothing happens

sebandraos, 2018-01-16:
I have sucessfully got librealsense working on the TX2 just this afternoon by following this [fork](https://github.com/syedharoonalam/buildJetsonTX2Kernel) of the jetsonhacks kernel. With that kernel built (I had to do a number of the operations manually rather than using the provided scripts) and a quick restart I was able to build librealsense 2.9.0 and run a number of the demos successfully. There are other issues (#840 ) but its a really good start.

dreamibor, 2018-01-22:
@sebandraos Hi, which Real Sense version are you using? Is that the Creative BlasterX Senz3D? Thanks

sebandraos, 2018-01-22:
The camera I am using is the SR300 per the question. The librealsense version is 2.9.0 which is the latest stable release at time of writing.

dreamibor, 2018-01-22:
@sebandraos Thanks for your answer. Are you using the pure SR300 module or like a product? I am considering to buy an SR300 based product called Creative BlasterX Senz3D, but the official document said it doesn't support Linux, so I an curious about which one are you using? Or can you suggest a website where you bought it? Thank you again.

mbayog02, 2018-02-21:
@sebandraos I have try the link you provided but I got the same error
Are you using SR300?
Could you explain all the steps you followed to get the SR300 Working on Jetson tx2?
Thanks

mbayog02, 2018-02-26:
@sebandraos I have followed your steps and I am not getting SR300 Working properly.
When I try cpp-config-ui sometimes the IR image blinks, others it get Stuck and the most of the times it closes with the error of my first post.
Is there any way to test if the "patch" was correctly apply?
What way are you using to test if the camera is working properly?

Thank you

RealSense-Customer-Engineering, 2018-02-27:
[Realsense Customer Engineering Team Comment] 
Hi @mbayog02

Several suggestions based on my configurations:
1. Use the latest BSP 28.2 instead of 28.1. This helps the stability issue.
2. Apply the realsense camera patches, especially for uvcvideo driver patch
3. Build librealsense 2.10.0 and run 'realsense-viewer' instead of 'cpp-config-ui' which is an old app

RealSense-Customer-Engineering, 2018-03-25:
[Realsense Customer Engineering Team Comment] 

If you still has the issue, please refer to below patch for the latest BSP (L4T 28.2).
Copy the patch file to ./scripts folder and run the patch: ./scripts/realsense-ubuntu-xenial-jetson-tx2.sh to patch the uvc modules.
https://github.com/freemanlo/librealsense/blob/master/scripts/patch-realsense-ubuntu-xenial-jetson-tx2.sh

closed 2018-03-25 by RealSense-Customer-Engineering