realsenseai/librealsense #937: D435 unable to stream depth and infrared simultaiouslly

sogartar, 2017-12-22:
| Camera Model | D435 | 
| Firmware Version | 05.08.14.00 | 
| Operating System & Version | Ubuntu 16.04.3 |
| Kernel Version (Linux Only)  | 4.10.0 |
| SDK Version | 2.8.3 |

The D435 device can not stream both depth and infrared at the same time. I can stream color&depth or color&infrared. It fails silently, the depth frames just don't come. I can not test that with another model as this is the only one I have. I don't know if it is a firmware issue.
Here is my not so short example to test that.
```
#include <boost/test/unit_test.hpp>
#include <iostream>
#include <librealsense2/rs.hpp>
#include <set>
#include <tuple>

template <class StreamProfilesIt>
StreamProfilesIt findVideoStreamProfile(StreamProfilesIt streamProfilesBegin,
                                   StreamProfilesIt streamProfilesEnd,
                                   rs2_stream streamType, int streamIdx,
                                   int fps, rs2_format format, int width,
                                   int height) {
  for (StreamProfilesIt it = streamProfilesBegin; it != streamProfilesEnd;
       ++it) {
    if (!it->template is<rs2::video_stream_profile>()) {
      continue;
    }
    auto s = it->template as<rs2::video_stream_profile>();
    if (s.stream_type() == streamType && s.stream_index() == streamIdx &&
        s.fps() == fps && s.format() == format && s.width() == width &&
        s.height() == height) {
      return it;
    }
  }

  return streamProfilesEnd;
}

BOOST_AUTO_TEST_CASE(StandaloneRealSenseInfraredDepthTest) {
  typedef int Width;
  typedef int Height;
  typedef int Fps;
  typedef int StreamIdx;
  typedef std::tuple<rs2_stream, StreamIdx, Fps, rs2_format, Width, Height>
      StreamId;

  std::set<StreamId> selectedStreamIds = {
      std::make_tuple(RS2_STREAM_DEPTH, 0, 30, RS2_FORMAT_Z16, 1280, 720),
      std::make_tuple(RS2_STREAM_INFRARED, 1, 30, RS2_FORMAT_Y8, 1280, 720)};

  // Get device and sensors
  rs2::context ctx;
  rs2::device_list devices = ctx.query_devices();
  BOOST_REQUIRE(devices.size() != 0);
  rs2::device device = devices[0];
  std::vector<rs2::sensor> sensors = device.query_sensors();

  // Find required stream profiles
  std::set<StreamId> foundStreamIds;
  std::vector<std::pair<rs2::sensor, std::vector<rs2::stream_profile>>>
      sensorStreams;
  for (auto &sensor : sensors) {
    auto streamProfiles = sensor.get_stream_profiles();
    std::vector<rs2::stream_profile> selectedProfiles;
    for (auto &streamId : selectedStreamIds) {
      auto streamProfilesIt = findVideoStreamProfile(
          streamProfiles.begin(), streamProfiles.end(), std::get<0>(streamId),
          std::get<1>(streamId), std::get<2>(streamId), std::get<3>(streamId),
          std::get<4>(streamId), std::get<5>(streamId));
      if (streamProfilesIt != streamProfiles.end()) {
        selectedProfiles.push_back(*streamProfilesIt);
        foundStreamIds.insert(streamId);
      }
    }
    if (!selectedProfiles.empty()) {
      sensorStreams.emplace_back(sensor, selectedProfiles);
    }
  }

  for (auto &streamId : selectedStreamIds) {
    if (foundStreamIds.find(streamId) == foundStreamIds.end()) {
      std::cerr << "Could not find required stream "
                << rs2_stream_to_string(std::get<0>(streamId)) << std::endl;
    }
  }
  BOOST_REQUIRE(foundStreamIds.size() == selectedStreamIds.size());

  bool gotDepthFrm = false;
  bool gotInfraredFrm = false;

  // Start streaming
  for (auto &sensorStreamsPair : sensorStreams) {
    sensorStreamsPair.first.open(sensorStreamsPair.second);
    sensorStreamsPair.first.start([&](const rs2::frame &frm) {
      if (frm.get_profile().stream_type() == RS2_STREAM_DEPTH) {
        if (!gotDepthFrm) {
          gotDepthFrm = true;
          std::cout << "got depth frame" << std::endl;
        }
      }
      if (frm.get_profile().stream_type() == RS2_STREAM_INFRARED) {
        if (!gotInfraredFrm) {
          gotInfraredFrm = true;
          std::cout << "got infrared frame" << std::endl;
        }
      }
    });
  }

  sleep(1);

  for (auto &sensorStreamsPair : sensorStreams) {
    sensorStreamsPair.first.stop();
    sensorStreamsPair.first.close();
  }

  BOOST_REQUIRE(gotInfraredFrm);
  BOOST_REQUIRE(gotDepthFrm);
}
```

dorodnic, 2017-12-22:
Hi @sogartar 
I think you need to refactor the code a bit, to use the overload of `open` that accepts a list of stream profiles ([rs_sensor.hpp:273](https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/hpp/rs_sensor.hpp#L273))
Please take a look at [unit-tests-live.cpp:120](https://github.com/IntelRealSense/librealsense/blob/master/unit-tests/unit-tests-live.cpp#L120) that does something similar. 
There is a firmware requirement for Depth and IR to be configured together, and sensor API tries to reflect it. We provide much simpler `pipeline` API that does all the stream profile matching internally (the implementation of `enable_stream` resolution is all localized in [config.h](https://github.com/IntelRealSense/librealsense/blob/master/src/config.h) if you want to check it out)
I'm not 100% sure why it is failing silently, you are supposed to get an exception on the second call to `open`. I will try to compile your code at the office and see what's up. 
Please let us know if any of this helped

dorodnic, 2017-12-25:
After compiling your code I'm getting:
```
got depth frame
got infrared frame
```
Do you still have this problem? 
Can you stream Depth + IR from this camera using the RealSense Viewer?

sogartar, 2018-01-02:
@dorodnic thanks for your support.
I tried to run on two other platforms and it worked there.
There may be something fishy with my USB 3 card. I will try with another USB card and if all is well will close the issue.

sogartar, 2018-01-02:
The problem was with the USB card.

closed 2018-01-02 by sogartar