realsenseai/librealsense #939: Question about MODIFIED_BROWN_CONRADY

freealong, 2017-12-24:
The implement of RS2_DISTORTION_MODIFIED_BROWN_CONRADY in rsutil.h:
```
if(intrin->model == RS2_DISTORTION_MODIFIED_BROWN_CONRADY)
{

        float r2  = x*x + y*y;
        float f = 1 + intrin->coeffs[0]*r2 + intrin->coeffs[1]*r2*r2 + intrin->coeffs[4]*r2*r2*r2;
        x *= f;
        y *= f;
        float dx = x + 2*intrin->coeffs[2]*x*y + intrin->coeffs[3]*(r2 + 2*x*x);
        float dy = y + 2*intrin->coeffs[3]*x*y + intrin->coeffs[2]*(r2 + 2*y*y);
        x = dx;
        y = dy;
}
```
According to the comment in rs_types.h,  RS2_DISTORTION_MODIFIED_BROWN_CONRADY is equivalent to Brown-Conrady distortion, except that tangential distortion is applied to radially distorted points. So should we update r2 before applying tangential?
```
if(intrin->model == RS2_DISTORTION_MODIFIED_BROWN_CONRADY)
{

        float r2  = x*x + y*y;
        float f = 1 + intrin->coeffs[0]*r2 + intrin->coeffs[1]*r2*r2 + intrin->coeffs[4]*r2*r2*r2;
        x *= f;
        y *= f;
        r2 = x*x + y*y; // update r2
        float dx = x + 2*intrin->coeffs[2]*x*y + intrin->coeffs[3]*(r2 + 2*x*x);
        float dy = y + 2*intrin->coeffs[3]*x*y + intrin->coeffs[2]*(r2 + 2*y*y);
        x = dx;
        y = dy;
}
```

RealSense-Customer-Engineering, 2018-05-17:
[Realsense Customer Engineering Team Comment]
hi @freealong,

sorry for no feedback, would like to know if you still need help on this topic?

freealong, 2018-05-17:
No need on this topic.

RealSense-Customer-Engineering, 2018-05-18:
[Realsense Customer Engineering Team Comment]
thanks and let me close this ticket.

closed 2018-05-18 by RealSense-Customer-Engineering