realsenseai/librealsense #940: python pyrealsense module has no align

alexanderPeregrina, 2017-12-25:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version |  3.21.0.0  | 
| Operating System & Version | Ubuntu 14.04 |
| Kernel Version (Linux Only)  |   4.4.0-98 |
| SDK Version | 2.8.1|

Hi, I'm traying to run the align-depth2color.py python wrapper example, but an error comes out saying:
align = rs.align(align_to)
AttributeError: 'module' object has no attribute 'align'
Does the align method already exist?

dorodnic, 2017-12-26:
Hi @alexanderPeregrina 
Looking at the [pull-request](https://github.com/IntelRealSense/librealsense/pull/776) for python align and releases page side by side, I see that align class was indeed not in 2.8.1. 
Can you try one of the later releases?

alexanderPeregrina, 2017-12-29:
Yes, that was the problem. I tried with the SDK 2.8.3, version, thanks.

closed 2017-12-29 by alexanderPeregrina