realsenseai/librealsense #941: Get Infrared images in Realsense SDK 2.0

lzppzlhotmail, 2017-12-25:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version |   | 
| Operating System & Version | Ubuntu 16.04  |
| Kernel Version (Linux Only)  | 4.10.0-42 |
| SDK Version | 2.0 |

I saw an example of Displaying Infrared Frame in getting_started_with_openCV.md.
But the example is based on Realsense SDK 1.0.
I found little info about the infrared in documents.
Could anyone tell me how about this feature in SDK 2.0?

Thanks,
Zhipeng

dorodnic, 2017-12-25:
Hi @lzppzlhotmail 
Are you refering to this version of [getting_started_with_openCV.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/stepbystep/getting_started_with_openCV.md#displaying-infrared-frame)?
It is accurate for SDK 2.0 as far as I can tell.
We also provide couple of extra OpenCV tutorials [here](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv) and a [Wiki](https://github.com/IntelRealSense/librealsense/wiki/API-How-To#start-streaming-left-and-right-infrared-imagers). 
The frame itself will have `frame.get_profile().stream_type()` equal to `RS2_STREAM_INFRARED`, and format `RS2_FORMAT_Y8`. It is an 8-bit grayscale image.

dorodnic, 2018-01-10:
Any further assistance is required on this issue?

lzppzlhotmail, 2018-01-12:
Thanks a lot!

closed 2018-01-12 by lzppzlhotmail