realsenseai/librealsense #946: Can't get more than 15 frames

Alonmeytal, 2017-12-25:
| Required Info  |   |
|---|---|
| Camera Model | D435 | 
| Firmware Version |  05.08.13.00 | 
| Operating System & Version | Win10 64bit 1709  |
| Kernel Version (Linux Only)  |  -  |
| SDK Version |  0.282.1 |

Running a simple loop to get a set amount of frames from the camera.
The loop constantly stops after 15 frames with a RuntimeError. The code is as follows:
```
for i in range(FRAMES_TO_RECORD):
		# Start measuring
		start_time = time.time()
		frame = pipeline.wait_for_frames(5000)
		frames.append(frame)
		time_now = time.time()
		print '[{}] {}'.format(time_now, i)
		time_per_frame_recieve.append(time.time() - start_time)	
```
It fails to get another frame even if I set the timeout to 20,000ms.

However, the Realsense viewer doesn't experience that issue.

dorodnic, 2017-12-25:
Hi @Alonmeytal 

This is happening because the script is saving frames inside a collection. 
At the moment, all frames are recycled and live within shared frame-pools. By default, we only allow for 16 frames to cycle in the system. 

This is done to reduce allocation time (we just fetch frame from the pool) and to ensure there is no consistent resource leak (if frames are processed slower then the rate at which they arrive, you must implement some kind of throttling mechanism or the application will run out of memory very fast). 

We do however acknowledge the requirement to be able to save unlimited number of frames (either for data-collection or in garbage-collected environments) and will enable it soon. For now, if you are building the library from source, you can build the `development` branch and replace `option_range{ 0, 32, 1, 16 }` with `option_range{ 0, 32, 1, 0 }` in [source.cpp:41](https://github.com/IntelRealSense/librealsense/blob/development/src/source.cpp#L41). After rebuilding (with `-DBUILD_PYTHON_BINDINGS=true`) the script should work.

Alonmeytal, 2017-12-25:
Thanks for elaborating, I’ll give it a try!

closed 2017-12-25 by Alonmeytal

tispratik, 2019-05-30:
Is there a workaround other than installing from source? I tried deallocating objects running gc in python, but none are helping.

guyoffe, 2020-05-10:
I solved this problem by closing the pipelines and starting them again.

PierreLouisT, 2022-01-19:
Hello ! 
I got the same issue. 
Im doing multiprocessing so I need to stock the frames in a buffer which will be use lately. However I can not get more than 15 frames (while k in the code). I understand the problem, Dorodnic explained it very well, thanks to him. Unfortunately, I am not developper and do not know how to solve it, it seems complicated. 
I am using only python and not C++.

```import multiprocessing as mp
import os
import threading
import time
import numpy as np
import pyrealsense2.pyrealsense2 as rs
import cv2
import sys
import math
current = (0,0)
class Camera(): 
    

    def __init__(self): 
       
        self.pipeline = rs.pipeline()
        self.config = rs.config()
        self.cfg = self.pipeline.start(self.config)
        self.dev = self.cfg.get_device()
        self.colorizer = rs.colorizer()

        #self.align_to = rs.stream.depth

    def param(self): 
        return self.pipeline, self.config, self.cfg, self.dev, self.colorizer

    
def on_mouse(event, x, y, buttons, params) : 
    global current

    if event == cv2.EVENT_MOUSEMOVE:
        
        current = (x, y)
             
    
def acquisition_frames(framelist, lock):
    print("Acquisition des frames")
    pipeline, config, cfg, dev, colorizer = Camera().param()
    print(dev.get_info(rs.camera_info.serial_number))
    print(dev.get_info(rs.camera_info.name))
    print(dev.get_info(rs.camera_info.firmware_version))
    print(dev.get_info(rs.camera_info.product_id))
    print(dev.get_info(rs.camera_info.physical_port))
    
    lst = []
    
    k = 0
    while k < 15 :
        lock.acquire()
        try : 

            frames = pipeline.wait_for_frames()
            depth_frame = frames.get_depth_frame()
            color_frame = frames.get_color_frame()
            depth_image = np.asanyarray(depth_frame.get_data())
            color_image = np.asanyarray(color_frame.get_data())
            depth_colormap = np.asanyarray(colorizer.colorize(depth_frame).get_data())
            lst.append(depth_colormap)
        finally : 
            lock.release()    
            k = k + 1
    framelist.put(lst)
   
if __name__ == '__main__':

    print("----------------------------------------------------------")  
    print("Main Process : définition des process")
    time.sleep(2)
    framelist = mp.Queue()
    lock = mp.Lock()
    process_acquisition = mp.Process(target = acquisition_frames, args = (framelist,lock,))
    print("Process  : lecture des données de la caméra et stockage des frames dans le buffer")
    start = time.time()
    process_acquisition.start()
    print("Main Process : Affichage du stream")
    print("----------------------------------------------------------")    
    getback_data = framelist.get()
    print("data bien récupérée")
    for i in range(15):
        cv2.imshow('RealSense depth colorized',getback_data[i])
        key = cv2.waitKey(200) # 1 frame toutes les x sec
    process_acquisition.join()
    end = time.time() - start
    print("temps realisé")
    print(end)
    print("fin ")   
    print("----------------------------------------------------------")

PierreLouisT, 2022-01-19:
Hello, I have the same problem : I need to surpass 15 frames because I am using multiprocessing, I read the commentary of @dorodnic which is very interesting and which explains well the problem. 
However, I am not very good at programming, can you help me to solve this ? 
I put my fonction below for processing : 
I want to stock all the frames I want inside a process called "acquisition_frames"
After that, I use this process inside a main fonction which get all the frames to disp the video flow

``` def acquisition_frames(framelist, lock):
    print("Acquisition des frames")
    pipeline, config, cfg, dev, colorizer = Camera().param()
    print(dev.get_info(rs.camera_info.serial_number))
    print(dev.get_info(rs.camera_info.name))
    print(dev.get_info(rs.camera_info.firmware_version))
    print(dev.get_info(rs.camera_info.product_id))
    print(dev.get_info(rs.camera_info.physical_port))
    
    lst = []
    
    k = 0
    while k < 15 :
        lock.acquire()
        try : 

            frames = pipeline.wait_for_frames()
            depth_frame = frames.get_depth_frame()
            color_frame = frames.get_color_frame()
            depth_image = np.asanyarray(depth_frame.get_data())
            color_image = np.asanyarray(color_frame.get_data())
            depth_colormap = np.asanyarray(colorizer.colorize(depth_frame).get_data())
            lst.append(depth_colormap)
        finally : 
            lock.release()    
            k = k + 1
    framelist.put(lst)