realsenseai/librealsense #947: Executing  rs-multicam crashes the operating system, ubuntu 16.04.3, kernel 4.10.

mjscosta, 2017-12-25:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | 3.21.0.0 | 
| Operating System & Version | Ubuntu 16.04.3 LTS |
| Kernel Version (Linux Only)  | 4.10.0-42-generic |
| SDK Version | librealsense2 |

Executing rs-multicam crashes the operating system.

[sr300_crash_report.log](https://github.com/IntelRealSense/librealsense/files/1586129/sr300_crash_report.log)

dorodnic, 2017-12-26:
Hi @mjscosta 
Is this a consistent crash? Does it reproduce only with multicam? How many cameras are you using?

mjscosta, 2017-12-26:
No, it is not consistent.

No, I had it happening twice, one following the other. First with realsense-viewer, the other wit rs-multicam.

Only one camera.

The strange, is that there is nothing in the kernel.log, syslog ... except for some strange characters.

The pc's hw:
Intel(R) Core(TM) i7-4600U CPU @ 2.10GHz
it is an hp EliteBook 820 g1.
all usb ports 3.0, only the cam is connected.

freemanlo, 2017-12-28:
My test configurations: 
1) Ubuntu 16.04.3 + Kernel 4.10.0-42, 
2) SR300 FW 3.21.0
3) librealsense 2.8.3
rs-multicam and realsense-viewer both apps have no trouble to run.  Probably, this is the system specific issue. Would you consider to try SR300 on the USB3 hub with an external power?

ev-mp, 2017-12-28:
Hi @mjscosta ,
The log is not very informative, but according to the description it is likely to be power-related issue.
Connecting the camera to the laptop via self-powered USB3 hub is the first thing I would do. Also make sure that the laptop is AC-powered when using SR300.
In case you have an additional camera, I would also cross-test it on this very laptop, while running the 'suspicious' SR300 on another, preferable desktop, PC to isolate the issue.

mjscosta, 2017-12-28:
Ok, thanks, I'll do that.

RealSense-Customer-Engineering, 2018-02-27:
[Realsense Customer Engineering Team Comment] 
Hi @mjscosta,

Is the issue resolved? Still need any help?

mjscosta, 2018-02-27:
You can close it, haven't seen it again.
Thanks for the support!

closed 2018-02-27 by RealSense-Customer-Engineering