realsenseai/librealsense #949: Is timestamp synchronization for fisheye and IMU supported?

David-Willo, 2017-12-26:
| Required Info  |   |
|---|---|
| Camera Model |  ZR300 | 
| Firmware Version | 2.0.71.28  | 
| Operating System & Version | Ubuntu 16.04  |

I try to capture the IMU , fisheye stream and color stream at the same time and record them into a ROS bag, but I find that the fisheye stream is slower than the color stream and IMU by a few seconds. Any advise?

[image]

icarpis, 2017-12-26:
Hello, @David-Willo, the sequence ID field in the [header](http://wiki.ros.org/Messages) of ROS messages doesn’t behave like a frame counter. This field is automatically increases as messages are sent from a given publisher.
Moreover, the sequence id will reset to 0 when you uncheck and recheck the topic on RViz. What probably happened is that there was a delay between the subscribe of depth topic to infra1 topic.

David-Willo, 2017-12-26:
@icarpis Thanks for your comment ! 
But what about the stamp field ? The fisheye stamp is 10 secs slower than imu stamp.
 I use zr300_nodelet_rgbd.launch from https://github.com/intel-ros/realsense to start my camera.

RealSense-Customer-Engineering, 2018-03-29:
[Realsense Customer Engineering Team Comment] 
@David-Willo Do you need further help for this?

David-Willo, 2018-04-02:
@RealSense-Customer-Engineering Yes! I really appreciate it!

RealSense-Customer-Engineering, 2018-06-22:
[Realsense Customer Engineering Team Comment]
Product has end of life'd and no fix is pending.

closed 2018-06-22 by RealSense-Customer-Engineering

maxwu8416, 2019-06-26:
hi~so is there anybody already solved this problem !?
i also have this problem and stop right here around a week....
hope someone can share your solution...