realsenseai/librealsense #952: Cannot load pyrealsense2 on Linux

Benz-Tracxpoint, 2017-12-27:
Linux, Kernel: 4.10.0-40-generic | Camera Model | D400 

Hi,
I am using RS D400 as part of intel's startups 
I am having trouble installing the development package on my LINUX
i will be more than happy for suggestions and helpful links as I went through the readme and wasnt able to use it.

I keep getting the following error message:
`  File "rs_tools.py", line 5, in <module>
    import pyrealsense2 as rs
ImportError: No module named pyrealsense2
`

PS - on windows the exact same python scripts is working

zivsha, 2017-12-28:
Hi @Benz-Tracxpoint ,
The error indicates that python cannot find the SDK's module named `pyrealsense2`.
The linux debian package does not contain the python wrapper module for the SDK.
You need to [build it from source](https://github.com/IntelRealSense/librealsense/tree/development/wrappers/python#installation-guide) at the moment.

Benz-Tracxpoint, 2017-12-31:
Thanks, it worked,
the only phase that was missing from our point of view was to run a make command after the cmake

closed 2017-12-31 by Benz-Tracxpoint

zivsha, 2017-12-31:
@Benz-Tracxpoint I will add this step to the instructions