realsenseai/librealsense #956: Creating source frames

jjoachim, 2017-12-28:
| Required Info  |   |
|---|---|
| Camera Model | D400 | 
| Firmware Version |   | 
| Operating System & Version | Ubuntu 16.04  |
| Kernel Version (Linux Only)  |   |
| SDK Version | 2613547178a35f... |

There doesn't seem to be a way to make frames myself; I can only grab them from the camera. It'd be nice if there was a way to deep copy and/or create frames from scratch so I can fill them in and use the given API for alignment and pointcloud generation. For instance, I'm interested in processing the RGB image to make a mask image, then align that to the depth frame to create a point cloud. The pointcloud's points would then be labeled by my mask image, and I can do further work from there.

It'd also be nice to be able to create a frameset from two frames to use in the align class api (for reasons stated previously).

Finally, "proccess" is spelled incorrectly various times in rs_processing.hpp (the file name is spelled correctly).

dorodnic, 2017-12-28:
Hi @jjoachim 
The requirement is completely reasonable, and we are looking into ways to enable this. 

At the moment the frame sub-system is trying to:
1. Provide thread-safe API for the user to manage the lifetime of individual frames 
2. Ensure fast and (more importantly) deterministic frame allocations when streaming

To accomplish both we have (a somewhat over-complicated) buffer-pool mechanism. That's why there is no API to create one single frame at the moment. 

One way to deal with this is to define a custom `processing_block`:
```cpp
processing_block pb([&](rs2::frame f, const rs2::frame_source& source)
{
    // source can be used to allocate new frames and send them out
});
frame_queue q;
pb.start(q); // Results from the block will be enqueued into q
pb.invoke(input_frame); // Invoke the lambda above
frame output_frame = q.wait_for_frame(); // Fetch the result
```
In this case, new frames will be allocated from the buffer-pool that exists inside the `processing_block` object. 
Unfortunately, this API is not well documented yet and isn't very user-friendly either. 

In addition, I am currently working on a "software only" device that can be used to generate frames from arbitrary data. It is still a [work-in-progress](https://github.com/dorodnic/librealsense/tree/bypass_dev), but will be most likely merged next week. The plan is to use such device to better unit-test point-cloud, sync and align methods, but one could also use it to inject data from an external source. Keeping your use case in mind, I will try to add a decent code-sample alongside this change.

Regarding spelling - the team is diverse and contains non-native English speakers (myself included). Fortunately, none of the misspellings are on the binary API boundary (C API), so we will fix it ASAP.
Thanks for pointing this out.

jjoachim, 2017-12-28:
Is there a similar processing block way of creating a frameset from a depth and rgb frame (by copy or ref is fine as long as I know which it is)? I'm digging through the code and found rs2_allocate_composite_frame(...), but there's a lot of indirection I'm having trouble following.

dorodnic, 2017-12-28:
Yes, I found an example here: [spatial-filter.cpp#85](https://github.com/IntelRealSense/librealsense/blob/0cc659d985af42f311b2c1c6aa9664d17133621d/src/proc/spatial-filter.cpp#L85)

jjoachim, 2018-01-08:
You mentioned merging your "software only" device most likely last week. Did this get merged? If so, could you point me to a code example and I think this issue can be closed.

dorodnic, 2018-01-08:
It is currently under review #966 (there is some extra functionality we need to add before it can be merged)

dorodnic, 2018-01-18:
Hi @jjoachim 
Please review [software-device example](https://github.com/IntelRealSense/librealsense/tree/development/examples/software-device) and [measure example](https://github.com/IntelRealSense/librealsense/tree/development/examples/measure) and let us know if this helps.

dorodnic, 2018-01-21:
Closing this one, feel free to open new issue if you have more questions

closed 2018-01-21 by dorodnic