realsenseai/librealsense #957: Sync issue with HW triggered external camera

kiran1228, 2017-12-28:
| Required Info  |   |
|---|---|
| Camera Model | RS430 | 
| Firmware Version |  5.8.14.0 | 
| Operating System & Version |  Linux 16.04 |
| Kernel Version (Linux Only)  |  4.10.0-40-generic |
| SDK Version | 2.0 |

We are using the RS430 camera to trigger an external color sensor but we are seeing a very obvious desync of the two frames that was not present with previous librealsense. I'm attaching a link to a video showing this issue: https://drive.google.com/file/d/1V-7rtC81VgTBjCb6cEdxiRYy668LNx1d/view?usp=sharing

We have tried to reduce the `DEFAULT_V4L2_FRAME_BUFFERS` to 1 and that helps with the de-sync but we are only getting 15fps from the cameras in that case.

svensht2, 2017-12-28:
@kiran1228 and I work on the same project
The last known version where this was working properly was 2.6.0, then we moved to 2.8.3 and this became broken.
I suspect this has something to do with this:
https://github.com/IntelRealSense/librealsense/commit/e77819ab0560985845bd7b4238b93a1febdccc08#diff-7b062492497246457e780537329788e3

RealSense-Customer-Engineering, 2018-04-18:
[Realsense Customer Engineering Team Comment] 
@kiran1228
is the issue still there? which way do you sync with external RGB camera? can you describe more? Thank you.