realsenseai/librealsense #958: How to infrared streaming on python pyrealsense2

alexanderPeregrina, 2017-12-29:
| Required Info  |   |
|---|---|
| Camera Model | SR300| 
| Firmware Version | 3.21.0.0  | 
| Operating System & Version |  Ubuntu 14.04 |
| Kernel Version (Linux Only)  |  4.4.0-104 |
| SDK Version | 2.8.3|

How can I get an infrared frame on pyrealsense2, which are the methods that I must use,?

zivsha, 2017-12-29:
This is probably the quickest way:
```py
import pyrealsense2 as rs
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.infrared)
pipeline.start(config)
frames = pipeline.wait_for_frames()
ir = frames.first(rs.stream.infrared) # also I think this is possible: ir = frames[0]
```

alexanderPeregrina, 2017-12-30:
Thanks, it works

closed 2017-12-30 by alexanderPeregrina