realsenseai/librealsense #962: RuntimeError: No device connected

Benz-Tracxpoint, 2017-12-31:
| Required Info  |   |
|---|---|
| Camera Model |D400 | 
| Firmware Version |   | 
| Operating System & Version |  Ubuntu 16.04  |
| Kernel Version (Linux Only)  |   |
| SDK Version |  2 |

Following Compilation of the SDK with its python Wrapper I try to run the following code:
```
    pipeline = rs.pipeline()
    config = rs.config()
    config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
    config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
    cfg = pipeline.start(config)
```
but the following error keeps showing even-though the device is connected:

```
  File "rs_tools.py", line 85, in <module>
    record_aligned_rgbd_to_disk_dirs(baseDir,True,True)
  File "rs_tools.py", line 20, in record_aligned_rgbd_to_disk_dirs
    cfg = pipeline.start(config)
RuntimeError: No device connected
```

As mentioned above, the Device is connected, and the same code worked for us on a windows operation system

any ideas what are we missing?

dorodnic, 2017-12-31:
Hi @Benz-Tracxpoint 
Please see the [https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q&A](troubleshooting WIKI). Does the system recognise the device (shows it under `lsusb | grep 8086`)?
In addition, please post the `dmesg` logs when connecting the device.

Benz-Tracxpoint, 2017-12-31:
Hן Doron,
Yes it recognizes it : 

> Bus 002 Device 003: ID 8086:0b07 Intel Corp.

Benz-Tracxpoint, 2017-12-31:
dmesg:

> [ 5239.077008] usb 2-1: USB disconnect, device number 3
> [ 5240.948783] usb 2-1: new SuperSpeed USB device number 4 using xhci_hcd
> [ 5240.969600] usb 2-1: New USB device found, idVendor=8086, idProduct=0b07
> [ 5240.969607] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [ 5240.969612] usb 2-1: Product: Intel(R) RealSense(TM) 430
> [ 5240.969616] usb 2-1: Manufacturer: Intel(R) RealSense(TM) 430
> [ 5240.969619] usb 2-1: SerialNumber: 725213020435

ev-mp, 2018-01-01:
Hi @Benz-Tracxpoint ,
Did you follow the [linux installation chapter](https://github.com/IntelRealSense/librealsense/blob/development/doc/installation.md) to set the required patches and permissions?
If so - please run 'rs-enumerate-devices' and post here.

Benz-Tracxpoint, 2018-01-01:
Hi,
I was finally able to get it work after following this tutorial as well:  https://software.intel.com/sites/products/realsense/sdk/getting_started.html#Setting_Up_Your_C_Development

closed 2018-01-01 by Benz-Tracxpoint

alexander-soare, 2022-08-02:
When I ran into this issue, a `sudo apt update` after the step of adding the librealsense server to the repo list did the trick.