realsenseai/librealsense #965: Noisy point cloud from R200.

CreFroD, 2018-01-01:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version | 1.0.72.10  | 
| Operating System & Version |  Ubuntu 16.04.3 LTS |
| Kernel Version (Linux Only)  |  GNU/Linux 4.4.0-104-generic x86_64 |
| SDK Version | 1.12.1 |

I use R200 camera with librealsense and PCL 1.8.0 in my project. I fill the point cloud with points from the camera with following code: 
```C++
int main() {
    pcl::PointCloud<pcl::PointXYZ>::Ptr cloud(new pcl::PointCloud<pcl::PointXYZ>(DEPTH_IMG_WIDTH, DEPTH_IMG_HEIGHT));
    //RealSense init
    rs::log_to_console(rs::log_severity::warn);
    //rs::log_to_file(rs::log_severity::debug, "librealsense.log");

    // Create a context object. This object owns the handles to all connected realsense devices.
    rs::context ctx;
    printf("There are %d connected RealSense devices.\n", ctx.get_device_count());
    if (ctx.get_device_count() == 0) return EXIT_FAILURE;

    rs::device *dev = ctx.get_device(0);
    printf("\nUsing device 0, an %s\n", dev->get_name());
    printf("    Serial number: %s\n", dev->get_serial());
    printf("    Firmware version: %s\n", dev->get_firmware_version());

    dev->enable_stream(rs::stream::depth, DEPTH_IMG_WIDTH, DEPTH_IMG_HEIGHT, rs::format::z16, 60);
    dev->start();

    while (true) {
        dev->wait_for_frames();

        const uint16_t *depthImage = (const uint16_t *) dev->get_frame_data(rs::stream::depth);
        rs::intrinsics depthIntrin = dev->get_stream_intrinsics(rs::stream::depth);
        float scale = dev->get_depth_scale();

        int dw = depthIntrin.width;
        int dh = depthIntrin.height;
        int dwh = dw * dh;

        // Set up the clouds to be used
        cloud->clear();

        cloud->width = DEPTH_IMG_WIDTH;
        cloud->height = DEPTH_IMG_HEIGHT;

        cloud->resize(dwh);

        cloud->is_dense = false;

        for (int dy = 0; dy < dh; dy++) {
            for (int dx = 0; dx < dw; dx++) {
                uint16_t depthValue = depthImage[dy * depthIntrin.width + dx];

                float depth_in_meters = depthValue * scale;

                if (depthValue == 0) continue;

                rs::float2 depthPixel = {(float) dx, (float) dy};
                rs::float3 depthPoint = depthIntrin.deproject(depthPixel, depth_in_meters);
                cloud->points[dy * dw + dx] = pcl::PointXYZ(depthPoint.x, depthPoint.y,
                                                            depthPoint.z);
            }
        }
        //Visualization and PCL processing...
    }
}
```
The point cloud that I get with this code is very noisy. Flat surfaces become "wavy":
[image]

When I try to calibrate the camera on windows with Intel's software, I get `DS_INVALID_ARG` error. Reinstalling Windows doesn't help. Thanks for help!

freemanlo, 2018-01-04:
Hi @CreFroD 
which Win10 version and the calibration tool do you use? Is below tool? Is DCM also installed?
https://downloadcenter.intel.com/download/24958/Intel-RealSense-Camera-Calibrator-for-Windows-

CreFroD, 2018-01-05:
Hello @freemanlo 
I use Calibrator 1.0.8 on Windows 10 Home Single Language, version 1709, OS Build 16299.15.

CreFroD, 2018-01-05:
Installation of older windows 10 (1607) helped with the calibrator, but it doesn't fix the problem. It says that my camera doesn't need calibration at all.

freemanlo, 2018-01-08:
You can still use the calibrator 1.0.8 to do below two ways and see how it goes.
1. force to do the caliration
2. reset to the factory settings
If it still can't be resolved, the factory calibration is required.

dorodnic, 2018-02-23:
Hi @CreFroD 
I closing this issue. If you need further assistance, feel free to open a new one or contact [RealSense support](https://www.intel.com/content/www/us/en/support/emerging-technologies/intel-realsense-technology.html).

closed 2018-02-23 by dorodnic